Klipper3d/klipper #2336: [Feature request] Neopixels on separate MCU with support for more LEDs

GarrethX, 2019-12-29:
Hey,

I've recently update Klipper to new version and wanted to try out the neopixel features. As I have chain with 25 pixels I came to an error, which limits me to maximum of 18 of them. However I was thinking, maybe it would be possible to bypass that limit while using a separate MCU? I know that neopixels are demanding when it comes to timing stuff, but adding a small arduino just for lightning could be a nice touch. 

Do you think thats feasible thing to implement? Or maybe it is already possible but I couldn't find it?
Thanks!

FHeilmann, 2019-12-29:
The limitation lies not in the MCU capabilities (although the old 8-bit AVRs struggle with neopixels on klipper) but in the message structure between the klippy host and the klipper MCU code. The message format only has room for 18 pixels. 

Therefore to fix this the neopixel code would need to be rewritten, depending on the processing power of your existing MCU, a separate one may not be necessary.

GarrethX, 2019-12-30:
Thanks for the explanation, but I'm missing something... If the limit is not the MCU itself, then we have two options:
1. Set the whole strip to particular color (ommit index parameter)
2. Set particular pixel to particular color (with index parameter).

If the limit is message, then I understand that everytime we set particular LED or whole strip, the message contains status of all pixels? From my point of view as an outsider, this seems a bit not optimal, as the MCU could keep in memory the previous status of all LEDs, and just get informed from host which LED to change. However I admit that I didn't dig deep enough in Klipper sources to understand how big deal is it, so please forgive me:)

FHeilmann, 2019-12-30:
Your approach certainly makes sense, in fact thats how most LED projects out in the wild do it. The reason it wasn't done this way here is that RAM is precious and klipper supports a large spread of devices where RAM availability differs greatly. It was thus decided to hold any data on the MCU, but instead use a '1 message => 1 LED strip update' mechanism, to keep it simple and not stress the devices too much.

GarrethX, 2019-12-30:
Ok, I understand:) So let's close this topic, maybe in the future things will change:)

Thanks!

closed 2019-12-30 by GarrethX