Klipper3d/klipper #2338: Documentation Improvement - bed levelling

gcormier, 2019-12-30:
After performing BED_MESH_CALIBRATE,

Will Klipper load any levelling mesh upon startup/restart?

I think the documentation could be clearer indicating if it is loaded or not by default, or if we should load at the start of gcode every time.

Arksine, 2019-12-31:
Hello @gcormier, there is new documentation that will be merged after the current feature freeze is up. Included in the documentation is a blurb on mesh profiles.  Each time you execute BED_MESH_CALIBRATE it temporarily saves the current mesh state to the *default* profile.  If you then issue a SAVE_CONFIG command, the default profile is saved to printer.cfg.  If a default profile exists it is automatically loaded at startup.

gcormier, 2019-12-31:
Super, that will make it more clear :) I'll leave it up to you if you want to close this issue or not.

Hywelmartin, 2020-01-02:
@Arksine is there a "clear_mesh_profile" command? or do you have to restart klipper?

gcormier, 2020-01-02:
@Hywelmartin Yes, there is. If you type HELP in the terminal for Klipper it will display the list of available commands including BED_MESH_CLEAR.

Arksine, 2020-01-02:
@gcormier is correct, BED_MESH_CLEAR will clear the mesh state.  If you want to delete a profile, use `BED_MESH_PROFILE REMOVE=profile_name`.

closed 2020-01-23 by KevinOConnor