Klipper3d/klipper #2339: Klipper state: Disconnect Printer is not ready The klippy host software is attempting to connect. Please retry in a few moments. Klipper state: Not ready

hydroculator, 2020-01-01:
I am trying to use klipper with a rPi 4, with an Ender 3.
I had klipper/Ender 3 working once with an Rpi3 before, but then upgraded to a Rpi4, and can't get it to connect. Once I updated klipper and started burning updated firmware, things went haywire.
   
I've reflashed both the bootloader and rebuilt and flashed the firmware multiple times.
The printer port is set to /tmp/printer in Octoprint. That part is up and running, and seems to be able to control klipper enough to yield the following logs in terminal.

05:02:31  Klipper state: Disconnect
05:02:56  Printer is not ready
The klippy host software is attempting to connect. Please retry in a few moments.
Klipper state: Not ready
05:02:56  Printer is not ready

It looks like the failure is just between Klipper and the printer itself.
When I flashed the printer, I use the gpio pins directly, and avr works as expected, so I believe the printer board is alive.

Attached is the klippy.log file.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4013819/klippy.log)

Thanks for any advice.

BlackStump, 2020-01-01:
did you check the changes here
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md
in particular this line.
>20191126: The USB names have changed on micro-controllers with native USB support. They now >use a unique chip id by default (where available). If an "mcu" config section uses a "serial" setting >that starts with "/dev/serial/by-id/" then it may be necessary to update the config. Run "ls >/dev/serial/by-id/*" in an ssh terminal to determine the new id.

hydroculator, 2020-01-01:
I did see that bit, and I did use what it returned, but that was while I was trying to flash the .hex files via usb. That never worked for me, so I used the gpio pin method which has always worked in the past.   
Would this change affect anything else in the configuration? I didn't see anything in the install instructions that would have applied once the images had been flashed. Did I miss a step there?

hydroculator, 2020-01-01:
FWIW, I use this guide to flash the E3 board from the rPi:

https://www.fission3d.com/guides/flash-bootloader-and-install-firmware-with-raspberry-pi

BlackStump, 2020-01-01:
You need to alter your printer.cfg with the results
your current printer.cg has
>[mcu]
>serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

Run "ls >/dev/serial/by-id/*" in an ssh terminal on your pi to determine the new id.

hydroculator, 2020-01-01:
I guess I skimmed by the result string and didn't notice they were different. Thanks for pointing that out.   This is the new string, which I've placed into the printer.cfg file.   
/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A107PNQF-if00-port0

However... 
I still get the same message.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4014349/klippy.log)

BlackStump, 2020-01-02:
try reflashing your firmware as per klipper instructions
># This file contains common pin mappings for the 2018 Creality
># Ender 3. To use this config, the firmware should be compiled for the
># AVR atmega1284p.

># Note, a number of Melzi boards are shipped with a bootloader that
># requires the following command to flash the board:
>#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
># If the above command does not work and "make flash" does not work
># then one may need to flash a bootloader to the board - see the
># Klipper docs/Bootloaders.md file for more information.

># requires the following command to flash the board: changed to match you serial id

avrdude -p atmega1284p -c arduino -b 57600 -P /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A107PNQF-if00-port0 -U out/klipper.elf.hex

KevinOConnor, 2020-01-23:
Any further updates on this ticket?

-Kevin

hydroculator, 2020-01-25:
> 
> try reflashing your firmware as per klipper instructions
> 

I finally replaced the board; nothing else worked.
It's just fine now.

closed 2020-02-02 by KevinOConnor

maxegor, 2021-01-25:
> > try reflashing your firmware as per klipper instructions
> 
> I finally replaced the board; nothing else worked.
> It's just fine now.

Have you solved the problem with the Ender 3 connection? What board did you change?