Klipper3d/klipper #2341: Print stop : "Stepper too far in past"

J-charles-C, 2020-01-01:
Hello and happy new year,

I have a problem. I did research without finding in the solution of others, one to my problem.

I enclose my log, my cfg and the g-code.

The failure occurs with each print and with different objects.

cordially

[Arrows.gcode.txt](https://github.com/KevinOConnor/klipper/files/4013923/Arrows.gcode.txt)
[printer.cfg.zip](https://github.com/KevinOConnor/klipper/files/4013924/printer.cfg.zip)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/4013925/octoprint.log)

BlackStump, 2020-01-01:
klippy.log is the log that is required as per above please read again and follow the guide to retrieve said log.

J-charles-C, 2020-01-01:
For several days, I have been reading kévin's answers and wondering how, from the octoprint log, he was able to understand the source of the problem.

Indeed, I did not read the good. I am a donkey.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4014315/klippy.log)

J-charles-C, 2020-01-02:
Ok, i stop. 
It's update problem. 
best

closed 2020-01-02 by J-charles-C