Klipper3d/klipper #2342: heated bed begins to heat indefinately at layer 2

EngineerBear8000, 2020-01-01:
Hi, 

Firstly thanks for creating this great project, ive been using klipper for my mishmash corexy (without heated bed) for the last few months and its been great.

I recently decided to finish installing my heated bed, did a pid calibration. All looked good, its able to heat up cool down and turn off as expected. But when i tried printing, it somehow locks on the heated bed on the second layer. kinda scary, I canceled the print before it reached 70c. Its definitely not my slicer settings, I did a word search through the gcode file for m140 and m190 and there are none, octoprint itself also reports the attempt to cool the bed, also the klippy.log below. The klippy_heat log shows it behaving normally when not printing so its also not the heater pin inverted, also tried inverting for the sake of completeness. 

Im stumped as to what could be the problem, hope u guys can give some advice. Thanks

[klippy_heat.log](https://github.com/KevinOConnor/klipper/files/4014073/klippy_heat.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4014074/klippy.log)

glowtape, 2020-01-01:
The klippy.log reports a target temperature of 0°C and responds to that with a 0% duty cycle on the PWM output to the heatbed MOSFET.

At the beginning it starts out at ~50°C, which was a test heating cycle? Then it slowly drops to 39°C while the toolhead is heating up and eventually the print starting. Then it slowly gets warmer. By 10°C over loosely 3 minutes. Doesn't quite read like a runaway heater.

Is the printer enclosed? How thick is the bed actually? Did the object to be printed have a large footprint and/or high infill density? Maybe it was absorbing heat of the actual print object?

EngineerBear8000, 2020-01-02:
Yes the 50c is from a previous print that I canceled while trying to find out how to get a log😅. 
Nope the printer is not enclosed.
It's a alu bed roughly 3mm thick with a spring steel sheet and insulation below. 
 
It did have a relatively large footprint but still less than 20% of the bed area it's a wrench with 15% fill I think. I'll find the gcode and put it here. It shouldn't be a heat capacity issue right? as the printer reports 0 degrees target

EngineerBear8000, 2020-01-02:
here was one of the gcode files i tried for reference 
[Uploading use-support-upperteeth.txt…]()

glowtape, 2020-01-02:
If you have a decent multimeter, you could arrange something to probe what the heater MOSFET outputs. If there's any voltage at all, and if so, if you have duty cycle measurement on the multimeter, what the cycle is.

The file upload didn't appear to finish before you submitted the reply.

EngineerBear8000, 2020-01-03:
[use-support-upperteeth.txt](https://github.com/KevinOConnor/klipper/files/4017670/use-support-upperteeth.txt)
oops 
ill  check with ghe dmm  tonight

glowtape, 2020-01-03:
The file you posted doesn't do anything odd with temps, from what I can tell. If the bed was running away, you ought to be able to measure an appropriate action on the MOSFET or on the leads to the bed.

If there's just a very thin coat of anything on 3mm of aluminium, like bluetape or something, and it's sort of isolated from the other side, say a PCB heater, I can imagine that it'll capture quite some of the heat from the plastic. As far as 70°C go, I'm not sure.

KevinOConnor, 2020-01-04:
The log files seem to indicate that Klipper is working as intended.  I do not know why the bed would gradually increase in temperature when the heater is not commanded on.  Some things to double check - make sure you have the correct thermistor for the bed (otherwise, the gradual increase may not be as significant as is being reported), and make sure the bed relay isn't inadvertently leaking some current into the bed heater.

-Kevin

EngineerBear8000, 2020-01-05:
unfortunately my dmm is unable to measure duty cycle, there are tho, multiple leds along the path to the bed. One on the ramps board one on my ssr and one on the bed itself. All light up solid after the second layer. however when its heating before the print starts the leds do pulse when reaching the target temp as expected from a pid controller.

BlackStump, 2020-01-05:
I would suggest you have a wiring/hardware problem and or a SSR problem, I do not think this is a klipper problem.
I would strongly suggest researching SSR connections and potential problems before using your heated bed. I would even suggest disconnecting your heatbed until you can verify the functionality of your SSR.
you may find this link useful
http://folgerforum.com/t/ssr-current-voltage-power-smoke-fire-how-to-heat-your-bed/947

EngineerBear8000, 2020-01-05:
found the issue, the mosfets on the ramps were bent and touching each other. shorted d8 and d9 thus the overriding of heated bed commands on second layer when the fan turns on. 
thanks for the advice

closed 2020-01-05 by EngineerBear8000