Klipper3d/klipper #2345: Quad Gantry Level Point Offset

archer183, 2020-01-02:
Just a feature request.
Add a touchoff offset for each corner in quad gantry level.    Quad gantry level does a great job of getting the bed back to the same degree of level each time, but what I have noticed is that the non contact proximity sensors (at least the ones I have used) can have some variance in trip point based on location on the bed.    

I have found it to be highly repeatable, in my case, front left and rear left trip about 0.1mm closer to the bed than front right and rear right.   As I can find no way to actually correct this out, I live with it, but simply being able to add an offset to each corner point would be a big help in fine tuning the machine

KevinOConnor, 2020-01-04:
FYI, this was raised in #2218 .  Also, you might want to double check the configuration of the motor locations, as a slight error in motor location can result in similar issues.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to close this in favour of the existing #2218 .

-Kevin

closed 2020-01-23 by KevinOConnor