Klipper3d/klipper #2346: Re-ARM with optical endstops

jalanjarosz, 2020-01-02:
I'm setting up a Re-ARM board with a new RAMPS board design and I'm having issues with optical endstops.

These are the optical end-stops I’m using: https://www.amazon.com/gp/product/B071KBC3P8
 
I have tried changing the trigger in .cfg (P1.24 versus !P1.24), using 3v3 and 5v, and they are not triggering.

3v3 trigger voltage is 0.2v (open) to 2.44v (triggered).  I've confirmed with Panucatt support that this voltage range should trigger the MCU pin and that the pin is also 5v tolerant.
 
When using a mechanical end-stop, all is working as expected.

I've tried to change the initial state of P1.24 in the make menuconfig to start as !P1.24 and I'm able to get a single trigger, but the state of the pin does not reset. when voltage is dropped.
 
I have gotten these same end-stops to work on a RAMPS board attached to an Adafruit Grand Central M4 MCU, which is alos a 3v3 logic MCU with no issues.

Please advise.
[klippy(1).log](https://github.com/KevinOConnor/klipper/files/4016660/klippy.1.log)

KevinOConnor, 2020-01-11:
FWIW, I personally, don't have any insight to what the cause would be.  It does sound like a hardware issue of some kind.

-Keivn

vladbabii, 2020-01-12:
Have you tried ^P1.24 ?

jalanjarosz, 2020-01-13:
@vladbabii Yes, I have tried inverting, pull-up, pull-down.  still no trigger or release of trigger.
@KevinOConnor  I suspect that you are correct.  I have engaged Panucatt and the RAMPS developer regarding the issue and we suspect it is due to having two sets of end stop filtering components in-line; one set built on the Re-ARM board and one set on the RAMPS board.     My next test is to jumper over one resistor on the RAMPS board and test again.

jalanjarosz, 2020-01-24:
Closing the ticket. 
I bypassed the noise filter on the RAMPS-32 and the optical end-stop functioned as expected.

I'll request the designer place a feature to bypass the on-board filter when using a Re-ARM MCU.

Thanks!

closed 2020-01-24 by jalanjarosz