Klipper3d/klipper #2347: How to Downgrade Klipper to V0.7

Achim01, 2020-01-02:
Hello,
after update to V0.8, I think the print quality is lower then before, with my Printer.
So I will try an earlier Version, but I do not know how.
Is there a simple way to do this. My knowledge of Linux is equal to zero.
Thanks.
J.W.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4017337/klippy.log)

robustini, 2020-01-04:
You will still need to set the MCU type, compile the firmware and flash it:

cd ~/klipper
git fetch
git checkout e4dac1039d82c7bcc7bdce20a7ba603eb117a905
sudo service klipper restart

Achim01, 2020-01-04:
The Downgrade was for test only.
I had problems right after the upgrade, but I think I found the Problem.
The Extruder does not feed regular, I changed everything except the Motor, with no result in v0.7.0 and v0.8.0.
I hope a new extruder will fix the problem.

Thanks.
J.W.

closed 2020-01-04 by Achim01