Klipper3d/klipper #2349: Constant errors, shutdowns and disconnects

patrick505, 2020-01-03:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4020900/klippy.zip)
I have been using Klipper for several weeks now based on many positive recommendations but I am so frustrated with it. It seems like almost any action I take causes the firmware to have an error and shut down. When it's cold, I have to heat up the machine with a heat gun to get Klipper to even start. Then I have to reboot the whole system. Anytime I try to move the tool head it shuts down. Every time I turn off the motors to level the bed I have to restart the firmware. Is there something wrong with my system? Or is there a way to auto restart the firmware after every action or disable shutdown upon every error? I have yet to see any significant improvement in prints since I mostly just been struggling to get it to start a print.

jakep82, 2020-01-03:
I don't see a log file anywhere in that attachment.  Please check the instructions at the link and post a log file for assistance.

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#issue-reporting

patrick505, 2020-01-03:
I checked the instructions but it didn’t specify where to find the log file so I copied the whole klippy folder. Let me know where to find it and I will attach it.

Patrick Reid
Media Services∙Lighting∙Repairs
tacticalvideoworks.com
505.699.9786

jakep82, 2020-01-03:
> Use the scp utility to copy the /tmp/klippy.log file from the Raspberry Pi to your desktop.

patrick505, 2020-01-03:
Where in here do I find that file?

Patrick Reid
Media Services∙Lighting∙Repairs
tacticalvideoworks.com
505.699.9786

jakep82, 2020-01-03:
It's in the /tmp directory and it's called klippy.log.  So the location is /tmp/klippy.log.  It is not in your home directory.

patrick505, 2020-01-03:
Do I need to “save” a log? How do I do that?

Patrick Reid
Media Services∙Lighting∙Repairs
tacticalvideoworks.com
505.699.9786

jakep82, 2020-01-03:
Please read the instructions at the link I provided.

patrick505, 2020-01-03:
Other than the M112 command, I see no other useful instructions. M112 still does not produce a tmp file.
What am I missing?

Patrick Reid
Media Services∙Lighting∙Repairs
tacticalvideoworks.com
505.699.9786

jakep82, 2020-01-03:
Are you able to SSH into your Pi?  If so, type the following commands:

`cd /tmp
ls`

Do you see a klippy.log file?

BlackStump, 2020-01-04:
The below is from the link above, not sure how one could miss or not read it. I would suggest that skimming the install instructions you have missed the how to setup Octoprint for klipper.

Issue requests are submitted through Github. All Github issues must include the full /tmp/klippy.log log file from the session that produced the event being reported. An "scp" and/or "sftp" utility is needed to acquire this log file. The "scp" utility comes standard with Linux and MacOS desktops. There are freely available scp utilities for other desktops (eg, WinSCP).

Use the scp utility to copy the /tmp/klippy.log file from the Raspberry Pi to your desktop.

Please make sure you have read and followed the install instructions fully. 
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

In particular this part for Octoprint
>In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but >stay connected to the printer" option. Click "Save".

KevinOConnor, 2020-01-23:
I'm going to close this issue as it looks like the original questions were answered (at least as best as we could).

-Kevin

closed 2020-01-23 by KevinOConnor