Klipper3d/klipper #2350: Ender 5: MCU 'mcu' shutdown: Rescheduled timer in the past

BryanHaven, 2020-01-04:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4021570/klippy.zip)

Encountered this tonight, was unable to issue the M112 immediately but was able to get in and get the logs, did a log extract and those are attached as well. 

Running an Ender 5 with the stock board + BLTouch. I've reset the pi and the printer, resliced and trying again. 

Any guidance would be appreciated. Thank you.

KevinOConnor, 2020-01-04:
Unfortunately, it looks like you got hit by a rare timing bug that was fixed a few weeks back (commit 988ee575).  I recommend upgrading to the latest version of the software.

-Kevin

BryanHaven, 2020-01-04:
Roger that, will do that after the current prints. Thank you!

BryanHaven, 2020-01-18:
Resolved with upgrade - no further incidents.

closed 2020-01-18 by BryanHaven