Klipper3d/klipper #2351: M106 improvements

robustini, 2020-01-05:
It is known that Klipper can use pins to control all the services that the user can add, but it would be very convenient to have a better one than the M106 command to control additional fans.
From what I see M106 does not recognize the "P" parameter, therefore "M106 P0 S255" and "M106 P1 S255" produce the same result, that is, they command the fan declared in "[fan]".
I have three fans declared in my config, but I would like to put one to cool the bed and control it with M106, possible?

[fan]
pin = ar9
kick_start_time = 0.200

[heater_fan extruder_cooler_fan]
pin = ar8
heater = extruder
heater_temp = 50.0
fan_speed = 1.0

[controller_fan board_fan]
pin = ar12
heater = extruder

It would be useful if Klipper could command additional fans without customizing the printer.cfg too much, simply by declaring a new fan:

[fan1] # (or something similar)
pin = ar6
kick_start_time = 0.200

So with M106 P0 I would control the fan on pin ar9, and with M106 P1 the latter on pin ar6.
Is an implementation of the M106 command possible?
It would be very convenient, thanks!
PS: the klipepr.log is a dummy file for the bot... :P

[klipper.log](https://github.com/KevinOConnor/klipper/files/4023023/klipper.log)

KevinOConnor, 2020-01-06:
FYI, this was discussed a bit in #2174.

I don't recommend making "enhancements" to traditional g-code commands (such as M106).  If someone wishes to implement and submit a more general SET_FAN_SPEED command then that's great.

-Kevin

KevinOConnor, 2020-01-23:
I'm going to close this as I think it is possible to implement this already using the method discussed in #2174 .

-Kevin

closed 2020-01-23 by KevinOConnor