Klipper3d/klipper #2352: 2x Ender 3 on the same raspberry

vsolenoid, 2020-01-05:
Hi all,

I have used 2 instances of octoprint on the same raspberry for my 2 Ender 3 with no issues. I wanted to test klipper on one of the machines, but now I'm unable to connect to the other, as the octoprint menu is missing the option to choose which port I want to connect to. 
I decided to install Klipper on the second Ender too, but when I select my micro-controller and write "make", Klipper returns "make: Nothing to be done for 'all'."
I saw then that whichever printer I connect to the raspberry the port is always /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AM009ADF-if00-port0

How should I proceed?

Thank you in advance

vsolenoid, 2020-01-05:
done

vladbabii, 2020-01-05:
You can modify /etc/systemd/system/klipper.service or /etc/init.d/klipper to add -I /tmp/printer2 to the second klipper instance. This would allow you to run both klipper instances at the same time

You can see other options here - https://github.com/KevinOConnor/klipper/blob/5eb5841eafccae5ca4b849647d39d7c5e3db9e64/klippy/klippy.py#L224

If you want to run multiple printer i would advise running also multiple octoprint instances

vsolenoid, 2020-01-05:
So I've followed this quite good tutorial https://www.reddit.com/r/3Dprinting/comments/d12dth/installation_guide_for_two_or_more_instances_of/ , but the result is still unfortunate. 
[klippy2.log](https://github.com/KevinOConnor/klipper/files/4023699/klippy2.log)
What I couldn't finish from the tutorial was sudo cp /var/run/klipper.pid /var/run/klipper2.pid 
Probably I had to make the PID test on the first klipper instance first?
Any other ideas?

Thanks

vladbabii, 2020-01-07:
You need a different pid file for each klipper instance. 
Would probably be easier to run it in docker (one docker with klipper and octoprint, for example) and run multiple instances of it at this point.

the  error
```
Lost communication with MCU 'mcu'
```
can have two main causes
- another software is trying to talk with the board (octoprint or something else) directly
- wiring is bad and usb/serial connection drops

try running 
```
dmesg
```

when connection fails and see if you have any usb errors

KevinOConnor, 2020-01-23:
Any further updates on this?  There is info on multiple instances in the FAQ - https://www.klipper3d.org/FAQ.html#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine .  Setting up the start/stop scripts is something that each individual user will need to figure out, however.

-Kevin

closed 2020-02-02 by KevinOConnor