Klipper3d/klipper #2356: G-code Macro parameter calls inconsistent

ld3300, 2020-01-07:
Okay, so I've been doing some more work with g-code macro parameters and have found that there is a handling inconsistency that it may be worth discussing.  Lets say I have:
```
[gcode_macro my_macro1]
default_parameter_S: 0
gcode:

[gcode_macro my_macro]
default_parameter_S: 0
gcode:
```
The issue is that if the name of the G-code macro ends in a number than the parameter call is done like a typical G-code command (i.e. my_macro1 S1) If the macro name does not end in a number than the parameter must use and equals sign (ie my_macro S=1).

So where this got a little more interesting is when I was setting up new dual extruder code for the recent T0, T1 changes (which are very helpful, thanks).

For this test I commented out my own T0 and T1 macros and took the T0, T1, PARK_extruder, and PARK_extruder1 example code from sample-idex.cfg as the base. I modified T0 and T1 as follows to test both use cases:
```
[gcode_macro T0]
gcode:
    ACTIVATE_EXTRUDER EXTRUDER=extruder
    TESTPARKa_{printer.toolhead.extruder} S=1
    TESTPARKb_{printer.toolhead.extruder} S1

[gcode_macro T1]
gcode:
    ACTIVATE_EXTRUDER EXTRUDER=extruder1
    TESTPARKa_{printer.toolhead.extruder} S=1
    TESTPARKb_{printer.toolhead.extruder} S1
```
The relevant parts of the TESTPARK macros:
```
[gcode_macro TESTPARKa_extruder]
default_parameter_S: 0
gcode:
    RESPOND TYPE=command MSG="EXT0 testparka = {S}"

[gcode_macro TESTPARKb_extruder]
default_parameter_S: 0
gcode:
    RESPOND TYPE=command MSG="EXT0 testparkb = {S}"

[gcode_macro TESTPARKa_extruder1]
default_parameter_S: 0
gcode:
    RESPOND TYPE=command MSG="EXT1 testparka = {S}"

[gcode_macro TESTPARKb_extruder1]
default_parameter_S: 0
gcode:
    RESPOND TYPE=command MSG="EXT1 testparkb = {S}"
```
The four states represented above result in the following:
TESTPARKa_extruder S=1 returns "EXT0 testparka = 1"
TESTPARKb_extruder S1 returns "Malformed command"
TESTPARKa_extruder1 S=1 returns "EXT1 testparka = =1"
TESTPARKb_extruder1 S1 returns "EXT1 testparkb = 1"

The result is that you are unable to use the same syntax on extended g-code commands depending on if they contain a number at the end of the name.  The {printer.toolhead.extruder} is just an example.  It is easy enough to work around, but I have had a few times where needing, or not needing an equals sign has thrown me.

I'm struggling to suggest a good solution.  Having a number at the end makes it behave like regular g-code.  Especially when implementing codes like M300 where you would expect it to behave like any other. And if the space between the code and the parameter is not present the g-code can still be processed?  Is the equals sign required when a number isn't present to distinguish the parameter value from the gcode macro name?  It seems counter to Klipper to require a number at the end of a macro name?  Is it possible to implement all extended g-codes without the equals sign, or find a way to have consistent behavior?

Log file is attached.  I did the T0 and T1 tcommands and issued a M112 immediately after the results.  

[klippy.log](https://github.com/KevinOConnor/klipper/files/4028536/klippy.log)

KevinOConnor, 2020-01-08:
This was a bug which I think should be fixed now (commit 005cfea5).  I also removed the limitation of using digits in "extended" g-code names.

Thanks,
-Kevin

KevinOConnor, 2020-01-08:
Upon a closer look, any numbers must still be at the end of a gcode_macro name.  However, I think the issue reported here should be fixed.

-Kevin

ld3300, 2020-01-08:
Here I was thinking I was being all helpful and detailed but you had already solved it.  Thanks Kevin.  I appreciate your work and wish I had the skills to contribute more.

closed 2020-01-08 by ld3300

KevinOConnor, 2020-01-08:
> Here I was thinking I was being all helpful and detailed but you had already solved it. 

The details were appreciated - I would not have found the bugs without them.

Thanks,
-Kevin