Klipper3d/klipper #2358: Klipper doesn't start.

Nibb31, 2020-01-07:
I've been using Klipper for over a year and I absolutely love the experience. I'm running it on the latest octopi on a raspbian stretch on a Raspberry Pi 2B over Ethernet with OctoKlipper. Octoprint runs fine, there are no errors in the octoprint.log. 

Since the last update, Klipper refuses to start. These are the commands I run to update Klipper:
```
sudo apt-get -y update
sudo apt-get -y dist-upgrade

cd /home/pi/klipper
git pull
sudo service klipper stop
/home/pi/klipper/scripts/install-octopi.sh
sudo make clean
make
sudo make flash FLASH_DEVICE=/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AC008T0F-if00-port0
sudo service klipper start
```
This script has worked perfectly in the past. It executes without any errors (except a warning about Python 2.7 being deprecated). The make flash completes without errors so the board is properly recognized by the raspberry.

However, klipper seems to exit immediately and doesn't even generate a /tmp/klippy.log. In octoprint, the Connect button doesn't do anything and the Klipper tab is greyed out.
```
pi@octopi:~ $ sudo service klipper status
● klipper.service - LSB: Klipper daemon
   Loaded: loaded (/etc/init.d/klipper; generated; vendor preset: enabled)
   Active: active (exited) since Tue 2020-01-07 20:08:56 CET; 15min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 2184 ExecStart=/etc/init.d/klipper start (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/klipper.service

Jan 07 20:08:56 octopi systemd[1]: Starting LSB: Klipper daemon...
Jan 07 20:08:56 octopi klipper[2184]: Starting klipper: klipper.
Jan 07 20:08:56 octopi systemd[1]: Started LSB: Klipper daemon.
```
I'm stumped.

Nibb31, 2020-01-07:
As I said, there is no klippy.log, because the klipper process seems to exit before it creates one.

jakep82, 2020-01-07:
I can't offer help to get you running, but I personally don't update packages on a functional printer.  I especially wouldn't use dist-upgrade which might might make changes to core software or upgrade the kernel.  For future updates, I would only do a git pull, run the Klipper install script, and flash the MCU if necessary.

Nibb31, 2020-01-07:
Ok, I deleted my klipper folder and reinstalled from scratch and it worked. So I guess we can close this.

closed 2020-01-07 by Nibb31

Announcement, 2021-07-22:
similar thing happened to me. it was fixed by simply doing `sudo service klipper stop` followed by `sudo service klipper start`, it was strange because the `sudo service klipper status` command seemed to show that the service was exited, but I guess it was still running and had to be stopped? I don't know, weird. anyways, hopefully someone is helped by this.