Klipper3d/klipper #2359: Shutdown after selecting tool in octoprint without homing first

Oranginator, 2020-01-08:
Has anybody an idea why I need to home my printer first before I choose extruder or extruder1 as tool so I can heat it up and load or unload it.
Same in cura, I need to do a G28 before it allows cura to heat up any extruder. A regular G28 is in the startscript of the Gcode, so the printer would home itself after it is heated up and starts to print. If I don't home first before hitting the print button or choosing any extruder a as tool I get an error and the printer shuts down to safe-state.

Is there anything I can do? With "old-style" extruder configuration it worked quite well, after updating printer.cfg to the macro style extruder activation I got these errors.

The error is: Must home axis first: 0.000 0.000 0.225 [0.000]

I'm on build v0.8.0-231-g3fb0b7e4
[klippy.log](https://github.com/KevinOConnor/klipper/files/4034184/klippy.log)

BeatSlayer, 2020-01-08:
Your problem seems to be triggered because your extruders have different z offsets.

KevinOConnor, 2020-01-09:
As indicated, your `T0` macro requests movement of the toolhead (`SET_GCODE_OFFSET Z=0.2 MOVE=1`), so it isn't valid to run that macro prior to homing.  You can either remove the movement from the macro, always home prior to running the macro, or update the macro to only move if the machine has been homed (something like `SET_GCODE_OFFSET Z=0.2 MOVE={% if 'z' in printer.toolhead.homed_axes %}1{% else %}0{% endif %}`).

-Kevin

Oranginator, 2020-01-09:
> As indicated, your `T0` macro requests movement of the toolhead (`SET_GCODE_OFFSET Z=0.2 MOVE=1`), so it isn't valid to run that macro prior to homing. You can either remove the movement from the [...]

Thank you! I'll try that.

closed 2020-01-23 by KevinOConnor