Klipper3d/klipper #2360: TMC2209 stepper driver: Unable to read tmc uart 'extruder' register IFCNT

Kronoshifter, 2020-01-08:
I keep getting this error and for the life of me, I can't figure out why. The odd thing is that it only does so on the extruder driver. I have four other TMC2208 stepper drivers configured with uart without a hitch.

I've tried this with at least 4 different stepper chips, the last of which has been the TMC2209. I've confirmed that each one has the pads correctly soldered, I've tried several different pins that I've confirmed work with the other tmc stepper drivers configured with uart. I've updated both klipper and octoprint to the latest version. Twice. The driver works completely fine when I comment out the [tmc2209 extruder] section. I've investigated every avenue I can think of and nothing, and I mean *nothing* has worked in the slightest. I'm at a complete and total loss as to what to do next. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4034283/klippy.log)

ld3300, 2020-01-09:
When you tried other stepper drivers had those drivers worked correctly in other motor positions?

Kronoshifter, 2020-01-09:
Yes, I've swapped around the drivers. No dice. I've tried everything short of buying a new board, which I'm coming close to doing.

KevinOConnor, 2020-01-09:
The most common reason for this type of failure is not having a complete connection between the PDN-UART pin of the tmc2209 chip and the micro-controller.  Some stepsticks wire the "uart" connection to different stepstick pins which can cause a lot of confusion.

Also, the tmc2209 allows one to select an address (0, 1, 2, or 3) based on the settings of other pins on the tmc2209.  Depending on the wiring of some other stepstick pins, it may be necessary to specify an address in the klipper configuration file.

-Kevin

Kronoshifter, 2020-01-09:
Assuming the pinout that came with the stepstick is correct, I've got it connected to the proper pin on the stepstick. Even so, I've tried all the top facing pins and none of them worked.

The uart_address should only apply if I have multiple stepsticks connected to the same microcontroller pin, correct? If so, I shouldn't have to worry about that because I have each stepstick connected to its own driver pin.

I saw a video in which it was recommended to bend or clip the bottom side of the pdn/uart pins when using it on an MKS GEN board, which I am. Could this potentially be a solution? Could the microcontroller be causing some sort of interference in that case? I'm only hesitant because my first set of TMC2208s that I tried to use here didn't have bottom-facing pins, as I modified the stepstick for uart myself.

EtteGit, 2020-01-09:
I had the same issue (ramps 1.6+ with re-arm 32bits). I had to bend the clock pin (just next to the pdn pin). Worked like a charm for me.

Kronoshifter, 2020-01-09:
Hopefully that'll solve it then. I'll give it a shot tonight. I'm annoyed that I couldn't find this anywhere except in a video, which I don't typically watch for this sort of thing.

Kronoshifter, 2020-01-10:
Bending the pins didn't work on the 2209, but I had a spare 2208 laying around, and it did work for that one. Not sure what the issue was specifically with the 2209, maybe I did need to mess with the uart_address option. I can do more testing on that tonight, but either way, I suppose this issue is resolved. I'll keep it open for a bit longer so I can test the effects of the uart_address option.

Kronoshifter, 2020-01-11:
Final update. I got the TMC2209 working by setting the uart_address option to 3. No idea why this would be, given the description of the option in the examples, but I'm not going to complain. Thanks for your help everyone, looks like this is resolved.

closed 2020-01-11 by Kronoshifter

DIYlover, 2021-04-13:
@Kronoshifter Thx for your tips for TMC2209! It works like a charm!

HermanGaviriaMarin, 2021-09-16:
Hey I just downloaded and installed klipper on my ender 3 v2 with a BTT Octopus Board v1.1 (huge boy with 8 steppers). Anyways, as I was doing my pre flight checks I ran into an error when trying to extrude material using the extrude button on octoprint. As soon as the button is pressed the whole thing goes into shutdown mode. I can confirm the heaters work and the other motor work fine. I have everything working except this. I also have 2209s, what I dont understand is this "setting the uart_address option to 3" you speak about.

Neofitou, 2021-09-18:
I'm also seeing the same error on an Ender with and Octopus board. But I'm not getting past the x stepper. 
I have configured for the stock end stops and even tried cutting the diag pin off the stepper. I will PM HermanG and collaborate.

darkytoothpaste, 2022-01-10:
Kronoshifter, I had the same problem as you and I realise what the problem is. I forgot to remove all the 3 MS jumpers in my MKS Gen 1.4. Interestingly, without removing the jumpers, setting to the UART address to 3 works as well.