Klipper3d/klipper #2361: Move exceeds maximum extrusion when using extra restart in retraction settings

daderaide, 2020-01-09:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4038002/klippy.log)

I know this error has been reported many times but I think this is an actual issue that is not S3D doing something wrong (maybe it's my settings being wrong though, I admit)

I am using S3D as my slicer and if I use "extra restart distance" in my retraction settings, klipper gives me the error "move exceeds maximum extrusion". I think this is incorrect. I have an extra restart distance of 0.2mm that has been working well for me for a long time, but this call causes this error since there is no X/Y move associated with the extrusion.

I have Klipper set up for pressure advance and have followed the recommendations described here: https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md

That doc does not mention that using extra restart distance is a problem. Is this a problem with Klipper, or is this a problem with my settings?

BlackStump, 2020-01-09:
my guess would be either use "extra restart distance" or use pressure advance do not use both.
Pressure Advance is in its own way is pushing more filament after a retract then you are doubling up with the S3D feature.
If you tune Pressure Advance correctly then "extra restart distance" after retraction should not be required.

daderaide, 2020-01-09:
In my understanding, pressure advance definitely doesn't add extra restart distance after a retraction that I know of. I would really like to keep using it if I can as it really helps with the small bit that oozes out of the nozzle especially after a longer travel move. If it's not compatible, then I would like to suggest that info be added to the doc I referenced in my OP.

BlackStump, 2020-01-09:
Have a read how pressure advance works and what it tries to achieve.
https://github.com/KevinOConnor/klipper/blob/master/docs/Kinematics.md#pressure-advance

you could increase your max_extrude_cross_section in your config
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L107

daderaide, 2020-01-09:
I'm familiar with how to disable the max extrude setting and on how pressure advance works. As I said before, pressure advance doesn't account for the small bit of filament that oozes out of the nozzle during long travel moves, which is why I like at add a small amount of extra restart distance. This helps my prints not have thin/under extrusion after long travel moves. I am currently running my machine with the max extrude setting set very high, but I'd really rather not disable one of the great safety features that Klipper has. Since this is a bug report (rather than a help forum), I'd like to know if this is actually a bug or if this is the intended functionality of the software. If this is the intended functionality, then I would like to suggest that this info be added to the document https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md so others know that this setting is not supported.

KevinOConnor, 2020-01-09:
> I am using S3D as my slicer and if I use "extra restart distance" in my retraction settings, klipper gives me the error "move exceeds maximum extrusion".

It is hard to tell from just the one error message, but my initial interpretation is that S3D is doing something highly questionable.  My recommendation would be to disable that option.  (Specifically, it would seem it is trying to move the "extra restart distance" filament into the nozzle while at the same time it is performing an XY move - unfortunately, it's not really valid to perform an extruder priming move at the same time as an XY move in g-code - the g-code speed is defined to be the XY speed and not the extruder speed, which causes havoc during interpretation of the request.)

> That doc does not mention that using extra restart distance is a problem.

This is something we can add to the documentation.  We rely on reports like this so that we know what needs to be documented.

> pressure advance doesn't account for the small bit of filament that oozes out of the nozzle during long travel moves

This is common and the recommended solution is to add a small retract distance (and/or wipe on retract).  It's described briefly at: https://www.klipper3d.org/Pressure_Advance.html

I'm not familiar with S3D, but "extra restart distance" does not sound like the standard "retract distance" and it doesn't sound like the option you want.

-Kevin

KevinOConnor, 2020-03-22:
I've updated the docs to recommend not using "extra retract distance" (commit 75438764).

-Kevin

closed 2020-03-22 by KevinOConnor