Klipper3d/klipper #2362: Delta_calibrate with x/y offset

tezdal, 2020-01-09:
My probe has an offset of the x and y axes.
^
|		p
|			
|			n
|
| ______________________>

printer.cfg
`[probe]
...
...
x_offset = -27.465695
y_offset = 16.154144
z_offset = 3.953`

When using the delta_calibrate
Printer does not come the probe to the calibration point, the nozzle comes to this point
As a result, the calibration is obtained not based on the center but with an offset in my coordinates

tezdal, 2020-01-09:
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4040334/klippy.log.zip)

tezdal, 2020-01-09:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/4041287/klippy.zip)
add klippy.log

KevinOConnor, 2020-01-09:
> Printer does not come the probe to the calibration point, the nozzle comes to this point

This is by design and is a requirement of delta calibration.

> My probe has an offset of the x and y axes.

FYI, I've never seen a success report for probes mounted on the side of the extruder on a delta.  Although these probes are common, they are susceptible to effector tilt which makes them unsuitable for the high precision needed during calibration.  I recommend you perform manual calibration as described at https://www.klipper3d.org/Delta_Calibrate.html

-Kevin

tezdal, 2020-01-10:
Ok, but how will the blatch work then?
It is also always with an offset, it is always next to the extruder and not under it. And it works great.

ProfDrYoMan, 2020-01-10:
You need a probe directly under the nozzle or the nozzle itself.

Everything else will simply not work due to the inverse kinematic of the delta.

Either something like that:
https://www.thingiverse.com/thing:2372733

Or that:
https://www.precisionpiezo.co.uk/

robustini, 2020-01-13:
I agree with @tezdal , the automatic calibration should be performed considering the probe x/y shift.
However, with my probe the automatic calibration went well enough, I only had to manually adjust the endstops a bit, but in size it is quite good.

tezdal, 2020-01-13:
@robustini I agree with you, the effector of my printer is fixed well. Why can't the Delta_Calibrate_with_Offset command be added?

KevinOConnor, 2020-01-23:
As noted above, the delta calibration logic does not use the probe XY offset and can not use it.  (Specifically, one can't use XY distances to calibrate a delta because on an uncalibrated delta the printer does not know how to move an accurate distance in an XY direction.)  If someone has a clever idea on this, they'll need to implement it and submit it.

-Kevin

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like the original question was answered.  (Delta calibration does not use the probe xy offset.)

-Kevin

closed 2020-02-14 by KevinOConnor