Klipper3d/klipper #2363: Embed filament sensor Error message

michaelwa01, 2020-01-09:
I have added the following lines so that I can install a filament sensor

[pause_resume]
recover_velocity: 50.

[filament_switch_sensor0]
pause_on_runout: True
event_delay: 3.0
pause_delay: 0.5
switch_pin: ^PA2

Now I get the following error

9:41:46
Section 'filament_switch_sensor0' is not a valid config section
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted
Klipper state: Not ready
19:41:46
Section 'filament_switch_sensor0' is not a valid config section

Maybe someone can help me how to do it right

I want to apologize for the bad English

Hywelmartin, 2020-01-09:
[filament_switch_sensor0]

[filament_switch_sensor my_sensor]

michaelwa01, 2020-01-09:
I have changed this unfortunately the same error, I have copied the settings into the configurations file, that's already correct or

Arksine, 2020-01-10:
@michaelwa01 Please attach klippy.log so we can help determine the source of your error.  Thanks.

michaelwa01, 2020-01-10:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4046509/klippy.log)

[printer.txt](https://github.com/KevinOConnor/klipper/files/4046540/printer.txt)

Hi there
I have now updated Klipper from 0.7 to 0.8 so far everything has worked, now I want to flash the MCU, unfortunately the error is now coming

It always worked :-(

I have an Ender3 with original MCU and rasperriy PI

It always worked great

Unfortunately I'm not so fit in English :-(

 just wanted to install a filament sensor :-(

pi@octopi:~/klipper $ avrdude -v -p atmega1284p -c arduino -P /dev/ttyUSB0 -b 115200 -D -U flash:w: /home/pi/klipper/out/klipper.elf.hex

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 5.0
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: reading input file ""
avrdude: error opening : No such file or directory
avrdude: can't determine file format for , specify explicitly
avrdude: read from file '' failed

avrdude: safemode: lfuse reads as 0
avrdude: safemode: hfuse reads as 0
avrdude: safemode: efuse reads as 0
avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

michaelwa01, 2020-01-10:
I think I got it right, the clipper is working again, I ask the following filament sensor:
BTT SFS V1.0 Smart Filament Sensor Detection

So you can also see a clogged nozzle, I hope you can integrate the sensor, if so what do I have to configure here?

Best regards Michael

Arksine, 2020-01-10:
Glad you got it working.  Unfortunately only switch based runout filament sensors are supported at the moment, so it isn't possible to configure the sensor you listed.

michaelwa01, 2020-01-10:
Hello thank you for the info, will there be such a thing in the future

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the original issue has been addressed.  Separately, I don't know of anyone currently working on the BTT SFS V1.0, but I know there has been general interest in filament sensors - only time will tell.

-Kevin

closed 2020-02-02 by KevinOConnor