Klipper3d/klipper #2366: Move exceeds maximum extrusion (8.259mm^2 vs. 0.640mm^2) with Simplify3D

eiten, 2020-01-10:
Hello there,

sometimes when using Simplify3D as a slicer, my prints stop with the message stated above. Coasting is disabled. Most of the time, it's on one of the first layers. I already contacted Simplify3D (as I suppose the problem lies there), but the payed support team was worth nothing, they told me they don't know Klipper but they are sure it's a Klipper problem :(.
Luckily, it's always in the first few layers.
Can you help me maybe? I attached the log and the g-Code. BTW: How can I see which line of g-code produced the error if I get something like this:
`Stats 157909.7: gcodein=29822420 mcu: mcu_awake=0.037 mcu_task_avg=0.000187 mcu_task_stddev=0.000185 bytes_write=67155957 bytes_read=12528735 bytes_retransmit=1623 bytes_invalid=0 send_seq=1295681 receive_seq=1295681 retransmit_seq=654031 srtt=0.009 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16000190 heater_bed: target=95 temp=95.1 pwm=0.633 print_time=157901.591 buffer_time=2.414 print_stall=5 extruder: target=235 temp=234.9 pwm=0.665
Move exceeds maximum extrusion (8.259mm^2 vs 0.640mm^2)
See the 'max_extrude_cross_section' config option for details
Unknown command:"M108"
Stats 157910.7: gcodein=29822608 mcu: mcu_awake=0.037 mcu_task_avg=0.000187 mcu_task_stddev=0.000185 bytes_write=67156662 bytes_read=12528910 bytes_retransmit=1623 bytes_invalid=0 send_seq=1295697 receive_seq=1295697 retransmit_seq=654031 srtt=0.010 rttvar=0.004 rto=0.027 ready_bytes=0 stalled_bytes=10 freq=16000190 heater_bed: target=95 temp=95.1 pwm=0.633 print_time=157902.391 buffer_time=2.214 print_stall=5 extruder: target=235 temp=234.9 pwm=0.659
`
Also something I don't understand: klippy.log says: Unknown command:"M108". But this is not in the gCode-file. Is this command triggerd by octoprint when the printer has an error?

Thanks and regards, Edi
[X-Carriage_Direct_Drive.txt](https://github.com/KevinOConnor/klipper/files/4046720/X-Carriage_Direct_Drive.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4046721/klippy.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/4046724/printer.txt)

eiten, 2020-01-10:
Oh sorry, I just saw #2361, somehow I did not stumble over it before. I'm using extra restart distance. I will try it again with the exact same settings except this parameter set to 0. I will report back (hopefully in 12 hours meaning that everything printed :))

Regards, Edi

eiten, 2020-01-10:
Nope. Exact same error :(

jakep82, 2020-01-10:
I've seen this error reported many, many times, and the common thread is always S3D.  I've also personally experienced it which is why I've stopped using S3D.  I would suggest trying Prusaslicer or Cura.

BlackStump, 2020-01-10:
I don't know if this helps but I have not had that error with S3D, check your setting choices under the 
g-code tab and change to relative E as per below example and retry.

[image]

eiten, 2020-01-10:
Well, I'd like to stay with S3D, I somehow like it. I will try the relative E settings. Thanks to both of you! I'll report back

krpepe, 2020-01-10:
@eiten I use S3D all the time and have never had an issue. What kind of printer are you using?

eiten, 2020-01-11:
I'm using an Ender 3. But does this matter? As the error occurs before the printer at Klipper?

krpepe, 2020-01-11:
@eiten I have a prusa clone if you want to compare config files to see if we can find the culprit. I wanted to make sure you had the same style of printer. Send me an email if you want to compare settings.
pepe3dprinting@gmail.com

Kevin

eiten, 2020-01-12:
@krpepe cool, thank you, I will.
@all does anybody know which gCode combination exactly causes this error? So maybe I could write a sanity checker?

BlackStump, 2020-01-12:
On a problem report the first step is to issue an M112 command in the OctoPrint terminal window immediately after the undesirable event occurs. This causes Klipper to go into a "shutdown state" and it will cause additional debugging information to be written to the log file.

BoHoehne, 2020-01-15:
I can confirm the problem for cura.

alfsoft, 2020-01-26:
I managed to get this error back in 2018 and since then I "temporary" solve this issue by altering extruder.py code as described [here](https://github.com/KevinOConnor/klipper/issues/461#issuecomment-418842623).
But after [this](https://github.com/KevinOConnor/klipper/pull/1997) commit my code no longer works so today after installing Klipper again I had to change part of the extruder.py code from this:
```
elif axis_r > self.max_extrude_ratio:
            if move.axes_d[3] <= self.nozzle_diameter * self.max_extrude_ratio:
                # Permit extrusion if amount extruded is tiny
                return
            area = axis_r * self.filament_area
            logging.debug("Overextrude: %s vs %s (area=%.3f dist=%.3f)",
                          axis_r, self.max_extrude_ratio, area, move.move_d)
            raise homing.EndstopError(
                "Move exceeds maximum extrusion (%.3fmm^2 vs %.3fmm^2)\n"
                "See the 'max_extrude_cross_section' config option for details"
                % (area, self.max_extrude_ratio * self.filament_area))
```
to this:
``` elif axis_r > self.max_extrude_ratio:
            if move.axes_d[3] <= self.nozzle_diameter * self.max_extrude_ratio:
                # Permit extrusion if amount extruded is tiny
                return
            area = axis_r * self.filament_area
            logging.debug("Overextrude: %s vs %s (area=%.3f dist=%.3f)",
                          axis_r, self.max_extrude_ratio, area, move.move_d)
            axis_r = self.max_extrude_ratio
            #raise homing.EndstopError(
            #    "Move exceeds maximum extrusion (%.3fmm^2 vs %.3fmm^2)\n"
            #    "See the 'max_extrude_cross_section' config option for details"
            #    % (area, self.max_extrude_ratio * self.filament_area))
```
Tested it on a few prints without any issue so you can also try it yourself, @eiten. Really not sure if this is the right way but my old altered code worked for a whole year on all of my print jobs.

KevinOConnor, 2020-02-14:
I'm going to close this issue as it looks like the conversation has concluded.  As far as we know, these requests from s3d are bogus and should be rejected by Klipper.

-Kevin

closed 2020-02-14 by KevinOConnor

cata747, 2020-06-27:
Hi. Just an idea about this. Today my bltouch mount losed a nut and my starting gcode had z-tilt-adjust and bed_mesh_calibrate in it.
The first print went horribly bad  and after that I was not able to start a print because of the "Move exceeds maximum extrusion" error. I use prusaslicer and not s3d.
After fixing the bltouch mount and saving another fresh config everything worked.
I was thinking that the wrong saved config profile of the bed was causing the error. I don't know. Just an idea.