Klipper3d/klipper #2370: Klipper turns off hotend after reaching target temp

BoHoehne, 2020-01-11:
Klipper keeps shutting down the hotend heater after reaching target temp.

The same hardware setup works with marlin.

[image]

[octoprint.txt](https://github.com/KevinOConnor/klipper/files/4049295/octoprint.txt)
[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4049296/klippy.log.zip)

I am new to Klipper. 
Any hint on how to solve this is welcome.

BoHoehne, 2020-01-11:
The origin of this behavior is CURA. 
This should be documented. 

I invested hundreds of hours in my cura profiles. 

Does anyone know what gcode is responsible ?

Hywelmartin, 2020-01-12:
Not sure that "R" is supported

```
Recv: B:63.1 /63.0 T0:205.8 /205.0
Recv: ok
Send: N12 M117 [          ]   5%*1
Recv: ok
Recv: ok
Send: N13 M73 P1 R39*28
Send: N14 M109 R205*91 <------
Recv: ok
Recv: ok
Send: N15 M107*17
```

Set extruder temperature and wait: M109 [T<index>] S<temperature>

    Note: M109 always waits for temperature to settle at requested value

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the original question was answered (don't use M109 R).

-Kevin

closed 2020-02-02 by KevinOConnor