Klipper3d/klipper #2371: Very random MCU disconnects.

p3dstore, 2020-01-11:
I have been fighting this issue for weeks, almost 2 months now on my Voron2.2. I've struggled to pinpoint it with the help of the Voron, Klipper, and Annex Engineering discord. I'll attach the latest log and my config. I'm running skr1.3's with [this](https://github.com/Stephan3/dwc2-for-klipper) flashed on the skrs for flashing. I've swapped USB cables, I've swapped pi's, I was on a 3b+ and now on a pi4b 1gb model. I've tried different klipper versions and it just keeps coming back. The error is damn hard to reproduce. Sometimes I can print fine for up to a week of 24/7 printing, or sometimes it fails on back to back print. I'll also attach the dmesg output from the pi. 

I am incredibly lost on the issue. I have another Voron running literally the exact same setup and no issues there. I'm currently thinking I have a bad skr or two, but I don't even know. The pi is powered from a meanwell rs-25-5 at 5.2v with a cut up usb-c cable. Please let me know what you would like to see. I have spare skr's on hand and I think I'll drop those in shortly. I need my printer running for my business, and this is quite the issue. I've wasted at least 5-10 spools of filament at this point. Let
 alone hundreds of hours wasted. 

I appreciate any help. Thanks!

[klippy log 1-11-20.zip](https://github.com/KevinOConnor/klipper/files/4049553/klippy.log.1-11-20.zip)

[dmesg1-11-19.txt](https://github.com/KevinOConnor/klipper/files/4049546/dmesg1-11-19.txt)

[Fade Config 1-6-20 pi4.txt](https://github.com/KevinOConnor/klipper/files/4049551/Fade.Config.1-6-20.pi4.txt)

KevinOConnor, 2020-01-12:
Ouch.

Unfortunately, something went terribly wrong on the host machine during the event.  You can see this in the dmesg log that you provided:
```
[150055.038239] octoprint invoked oom-killer: gfp_mask=0x6080c0(GFP_KERNEL|__GFP                                                                                                                        _ZERO), nodemask=(null), order=0, oom_score_adj=0
```

FYI, the "oom-killer" is a drastic response by the Linux kernel level to handle an out-of-memory event.

Around this same time, the Klipper logs show that the Klipper host software is unable to get cpu processing time (the stats messages which are normally reported once a second have gaps of 4 seconds then 10 seconds).  Eventually the micro-controllers respond as designed and go into a safety state.  Your description above is consistent with a transition to this safety state.  All of these responses are also consistent with a kernel level OOM response.

If there's an upside - there doesn't appear to be any indication of communication, usb, or micro-controller failures.

It's difficult to guess what caused the OOM event.  The obvious things to check: make sure you're on the latest version of octoprint, disable all octoprint plugins, try disabling any video cameras (if applicable).  You may also want to verify you have a recent (and high quality) sd card in the raspberry pi, and it may be worth a shot reinstalling Octopi on the sd card.  (Though I'll stress these are just random guesses.)

-Kevin

glowtape, 2020-01-12:
The OOM reaper killed a component of the GStreamer framework, that ate 880MB of RAM. Definitely webcam related.

p3dstore, 2020-01-13:
I'll update here if I get the errors. I'm running a complete clean install now. So time will tell.

p3dstore, 2020-01-19:
Seems to have been my pi overheating slightly maybe. Haven't had the dissconnects since going pi4 with active cooling.

closed 2020-01-19 by p3dstore