Klipper3d/klipper #2372: Add explanation to interrpolation in docs/Bed_Mesh.md

Coffee0297, 2020-01-11:
i need some more description in the Bed_Mesh.md. it describes a lagrange and bicubic, but not when to use something or other. see example why this is needed to document.

Lagrange  9 points : 
[image]

Bicubic 9 points : 
[image]

i have made some in my own branch that you can use. : https://github.com/Coffee0297/klipper/blob/master/docs/Bed_mesh.md

Arksine, 2020-01-11:
The newest version of bed_mesh will not allow lagrange interpolation for more than 6 samples along an axis as outlined in the Bed_Mesh.md:

>algorithm: lagrange
Default Value: lagrange
The algorithm used to interpolate the mesh. May be lagrange or bicubic. Lagrange interpolation is capped at 6 probed points as oscillation tends to occur with a larger number of samples. Bicubic interpolation requires a minimum of 4 probed points along each axis, if less than 4 points are specified then lagrange sampling is forced. If mesh_pps is set to 0 then this value is ignored as no mesh interpolation is done.

In my testing lagrange interpolation did not oscillate at 6 points or less, although its it may be possible if the points vary a great deal.  I'm not opposed to creating an appendix going into more detail on the interpolation algorithms.

Coffee0297, 2020-01-11:
@Arksine that sounds great. but what is the error msg when some tries to use above? do they get notified or do you just convert to bicubic in code?

ld3300, 2020-01-11:
You get a error that you can't have more than 6 points

Arksine, 2020-01-12:
As mentioned above, when the configuration is parsed a config error is raised notifying them that lagrange does not support more than 6 points.  The user has two options then, lower the probe_count or switch to bicubic interpolation.

Coffee0297, 2020-01-12:
thats sweet thanks, closing then.

closed 2020-01-12 by Coffee0297