Klipper3d/klipper #2373: Unusual Pressure Advance behaviour

Asterchades, 2020-01-11:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4049682/klippy.log)

I'm including the log merely as a formality, as I don't expect it will help. However, I'm also including pictures which... well, they may not help either. Allow me to explain.

Two days ago now, I ran off a new Benchy to test my latest profile modifications. I'm not a huge fan of the model but it's popular enough that comparing results (and issues) isn't usually a hassle. But I found that while _most_ of mine printed near perfectly, the pillars on the cabin (and a few other small details) were significantly under-extruded - to the point where I accidentally ripped off the top of the cabin. I switched out my extruder and tried again, only to find the same basic problems (albeit slightly lessened). So I printed off the model starting from 18mm up, compared the more-detailed front part of the cabin, and, well...

[image]
[image]
On the left is the cabin piece using my normal 0.82 Pressure Advance value. On the right is the same cabin piece (stopped a little later) with 0.0 Pressure Advance.

Figuring there was some problem with Pressure Advance in combination with short extrusions and multiple retractions on the same layer, I created a custom model with the same basic features to use as a test tower for tuning. This gave the following result...
[image]
Seen as how this now worked, I figured a setting had inadvertently become "stuck" or otherwise corrupted, so I printed off the earlier cabin piece again.
[image]
[image]
Same deal - PA 0.82 on the left, PA 0.0 on the right. PA definitely gave a better showing this time (especially from the front), but something is clearly off.

All of these cabin parts were printed on the same printer, off the same roll of filament, using the same G-Code made by Cura. The only thing that was changed was the PA setting. This issue also only appears to show up on layers with short extrusion/retraction combinations but doesn't apply to the whole layer - the hull of an affected print is absolutely fine, even up above where the cabin starts failing, until it reaches the section with the anchor holes in it.

For reference, here's a complete Benchy I ran off a couple weeks ago. It does have some minor surface problems but I don't feel they're significant enough to represent a problem. Importantly, though, the top of the cabin is firmly attached. The steering wheel is also perfectly formed (and near-impossible to photograph with my available equipment).
[image]

KevinOConnor, 2020-01-12:
Interesting.  Thanks for reporting.

What acceleration did you print this at?  What was the retract setting and retraction speed (if any) for the print?  When was the last time you calibrated pressure advance using the "square tuning tower"?  If you still have the tuning tower available can you post pictures of that?  Do you know the code revision you were at when you printed the benchy from a couple of weeks ago?

The most obvious concern would be something due to the merge of PR #2275 a few days ago.  However, a presure_advance=0.85 is pretty high and it's also possible the extruder is reacting poorly with an accel of 3000 (the default for your config).

-Kevin

Lingomat, 2020-01-12:
Suggestion: Print hotter and see if it gets better.

I experienced something like this issue, also a bowden and similar PA levels. Given the inherent friction of the filament in the path and nozzle, rapid back and forth extruder movements can 'scuff' the filament so the extruder gears fail to find purchase. By raising the nozzle temperature, it gets a bit easier to push, and the problem should subside.

In my case, this effect clued me into the fact I was printing too cold because I had the thermistor misconfigured. Might not be your problem, but it took me days to figure this out so just on the offchance it helps you...

Asterchades, 2020-01-12:
I'll try to hit each point in order... hopefully I won't miss anything.

Acceleration should have been the same 3,000 as listed in the config. The only changes I've made to the config of late have been adjustments for a recently-fitted BLTouch clone and changing the extruder step value when I switched the extruders for this test.

_On the subject of the extruder step value, I also had to invert the direction pin. Immediate testing after that actually had the motor fighting against itself as soon as I told it to move, even after any programmed movement had stopped. I had to restart the whole system (Pi and printer) before it would move freely again at all times._

Retraction is set to 2.5mm at 50mm/s with no minimum extrusion length between retractions.

It's been months since I calibrated my PA. I couldn't tell you exactly how long. I did more recently try using the Linear Advance Calibration Pattern (http://marlinfw.org/tools/lin_advance/k-factor.html) out of curiosity to see if it was a quicker way to test (I set up a macro for M900) but manually re-tweaked the value over a few prints back to where I started based on the actual results.

Oddly enough, I found my tower again just recently. I thought I had thrown it out, but I guess not.
[image]
Same filament is used for this as the previous pictures off the same spool. That "scar" on the right is also on one more side, and there's some downright weirdness happening in the middle (not pictured was the removed strings from the front left corner to the front of the right inner gap), but I put that down to some Cura settings I was playing with at the time. It's safe to say that even a value of 1.0 isn't perfect, but some corner inconsistency starts creeping in too far above 0.8.

Unfortunately I've no idea what exact version the pictured complete Benchy was using.

That PR was my concern as well. But I'd already tested the S-curve branch and bezeir fork, which had that rolled into it, without seeing any such problem - hence my confusion. I mean, I guess it's possible that I didn't test the right type of print to see it.

Regarding temperature, I already print at 210°c for this material (it's a PLA+ from 3DFillies in Australia). Ideally I'd like to print at 205°c for slightly improved overhangs as per a temperature tower, but that starts introducing this weird stringing that no amount of retraction solves. For posterity, though, I have tried again at 220°c (same G-code as before):
[image]
New print on the right is definitely better than the original on the left, but still not quite as good as the no-PA one in the middle. So this could be an avenue worth exploring further (maybe 225°c?)

In the short term, I'm going to try a few more things. I've got a few more parts incoming still (plated copper nozzle, clone BMG extruder, and all-metal hotend from Trianglelab) which may or may not influence it, but I'll try a few software options as well - lower acceleration (2,000mm/s/s) and higher temperatures. I'll let you know if it produces any relevant changes.

_Incidentally, since I didn't specifically mention it, I did try undoing all my recent profile changes just in case to no real effect. I didn't expect it would, though - I'd changed things like minimum polygon circumference, wiping and bridge settings._

Asterchades, 2020-01-12:
Well... this is awkward.

[image]

Left is the original. Middle is no PA. Right is what just came off the bed - PA 0.82, 210°c. Same G-code again save for the temperature modification.

I say it's awkward because I was a terrible scientist in this case. I made two changes at once, rather than making two separate changes in isolation. I both lowered the acceleration to 2,000mm/s/s _and_ completely re-installed Klipper. By "completely re-installed" I mean I ran the uninstall script, removed the folders, re-cloned the git, re-ran the installer, remade and reuploaded the firmware, and rebooted.

Plus side, though, I guess it means the new PA system wasn't the problem.

As to whether it was the acceleration or some kind of corruption/taint, well... I'm not game to change the acceleration while it's working, and I can't exactly attempt to re-corrupt anything since I've no idea how or why that might have happened (could be the memory card is at fault? Dunno...).

closed 2020-01-12 by Asterchades

Asterchades, 2020-01-12:
OK, so, like anyone with half a brain would have realised... not testing 3,000mm/s/s again is just slack. Can't come that far and miss the final element. So I went back and did it, just in case there's some weird incompatibility.

[image]
2,000mm/s/s on the left, 3,000mm/s/s on the right. Aside from a slight change in the vertical rippling (it's a cantilever gantry on POM rollers w/ bowden tube, so I expect that) they're basically identical. Which means it's not an acceleration incompatibility.

Of course, that still doesn't explain exactly what did happen. But it does give me a diagnostic step for the future - start by doing a full re-install. then take steps if the problem persists.

Apologies for any time wasted investigating.

jalanjarosz, 2020-01-12:
@Asterchades I’ve been following your issue and I’ve seen some similarities in my CoreXY.  
My printer with a BMG clone and 850mm Bowden tested with a high PA and retraction and still some issues with little blobs, stringing, and same oddities in small move/retract areas. 

I just did a Klipper-uninstall, rename the Klipper directories and finally a clean git clone. I’ll be doing another PA test later today. But the little print I’m running now seems quieter at the extruder.

Asterchades, 2020-01-12:
Well, if I've managed to help one person then it's not a waste.

Curious, though, if you're seeing similar issues it's unlikely to be corruption. Perhaps it's some kind of transitional artefact caused by updating at exactly the wrong moment, or between the wrong commits? Or just a really weird coincidence.

Leviathan220, 2020-02-27:
Hey just wanted to chime in here. Not sure if the issue should be re-opened or not. However, ever since I initiated the pressure advance test it's been stuck on. I have tried RESTART, FIRMWARE_RESTART, ssh into the pi and running sudo service klipper restart, and manually setting pressure_advance: 0.0 in the extruder config. None of this has worked. It is stuck on. I'm not sure why? I ran the klipper-uninstall.sh script. I reinstalled klipper. I reflashed my MCU. I used a brand new config file. It's still trying to pressure advance after the first layer? I'm at a loss.

Leviathan220, 2020-02-27:
Litterally trying to tune PA according to the official documentation has rendered my printer unusable.

Leviathan220, 2020-02-27:
Update: I have used the PRESSURE_ADVANCE ADVANCE=0.0 SMOOTH_TIME=0.0 command I have power cycled the printer and the raspberry pi.
Now my extruder tries to PA constantly even when running a simple load filament command. All I hear is the motor skipping as it turns in the opposite direction every few seconds. Nothing comes out of the hotend when printing. I checked the hotend is not clogged. I'm not sure why any of these setting persisted across a reinstall.

Asterchades, 2020-02-27:
Gonna go out on a limb and say that the two issues aren't related. I never experienced any trouble with it during things like a loading move (I use a 350mm load due to the bowden tube) - only a weird issue with underextrusion during actual prints, and only on very small layer sections (which may or may not have been related to acceleration as well).

I couldn't even begin to guess as to what might be causing your issue.

Leviathan220, 2020-02-27:
I don't think they are anymore. They only started after I tried to set up linear advance. Simply loading and unloading the filament from the lcd screen causes massive retracts? These retraction seem to be related to speed. The faster the extruder moves the more frequently it tries to retract. This extruder movement is will not go away after cleanly installing every component including a swap of SD cards on the pi just in case maybe that was the issue.

Leviathan220, 2020-02-27:
I have opened another issue for it. It started only after enabling PA. That's why I thought it might be related.

Camaro1, 2020-04-01:
Sorry for adding to a closed issue, but I had exactly the same issue. I am also printing with a very high pressure advance value of 0.85 due to the long bowden tube. Printed a Benchy and the pillars of the cabin show these artefacts whereas the hull of the ship on the same layer does not show and inconsistency.
One thing i changed from the stock settings was the pressure_advance_smooth_time. I set it to 0.100 for the extruder to move slower. I guess this causes the issue. I lowered the value to 0.02 which solved the issue. There are no more wavy surfaces. This might also explain why a new install solved this issue for @Asterchades. Maybe this also reset the smooth time value.
Now the extruder twitches sometimes, but I will have to observe if this effects the extruded plastic or if this is simply the sound it makes when extruding/moving very fast.
Maybe this helps someone with similar issues.