Klipper3d/klipper #2376: How to determine cause of serial connect failure?

xC0000005, 2020-01-12:
I'm running Klipper (or attempting to) on a Chitu-F board, which has an STM32F103 processor. I tacked on an option for a 34k bootloader (which is what Chitu uses), and can build, encrypt, and flash the firmware onto the board.  So far, so good, BUT, the board doesn't do USB communication in what I'd call a normal way. It has a CH340 chip connected to Serial1. I can connect perfectly well at 115200 baud and read the bootloader messages.

Klipper fails to connect. In order to troubleshoot, I hooked a Serial->USB converter where the ESP8266 is usually hooked for WIFI, and hooked that to the PI. Now the klipper log says it's connecting but times out on the identity response.

My build options are STM32, 34kb offset, 12mhz crystal (which matches what is on the board), USB disabled, Serial3 selected, baud rate set to 115200 (this is a holdover from me trying to use Serial1, which pretty much will only communicate at this rate - I can build test sketches in arduino, encrypt them, and load them on the board.

I've attached the klipper log. Is there a way to see more about what Klipper is getting from the MCU? At some point, I want to switch back to using the on-board USB via the CH340 and Serial1, but I can at least work on the pinout and config using the FTDI clone if I can figure out why it won't connect.

[Klipper.zip](https://github.com/KevinOConnor/klipper/files/4050119/Klipper.zip)

xC0000005, 2020-01-13:
Closing for now, until I can run a few test sketches. When I'm 100% sure what config and parameters ought to work, I'll switch back to using Klipper with those settings.

closed 2020-01-13 by xC0000005

ondra, 2020-04-08:
@xC0000005 , could you please share your config and the way of encrypting the firmware? I have a printer with the same board, this would save me lot of trial and error..

xC0000005, 2020-04-14:
My recommendation is to dump the board using the Cube programmer, and then flash on klipper at the base offset. Why do I recommend this? Because it works. Chitu consumed the SWD pins (one goes to the enable of the winbound, another is to the LCD) and so debugggin why it is that klipper won’t start with the bootloader enabled isn’t trivial. I’ll ifgure it out eventually but for now, just build a regular 103 binary with no offset and use the cube programmer to write it to the default address. 

My branch https://github.com/xC0000005/klipper/tree/resyncWithMaster <https://github.com/xC0000005/klipper/tree/resyncWithMaster> has the config I’ve been working on (occasionally) and my master branch has the scripts + makefile to encrypt for chitu and lerdge (but like I said, for now, just use the cube mx to write it in at 0x8000000).

ondra, 2020-04-14:
Thanks for the info -- turns out that I actually have a similar board, but different (X5SA Pro). The CPU and bootloader are the same, but pin map differs. Still no SWD, though.

I couldn't get it working with the bootloader -- as you say. The board flashes and runs _something_, but not enough to get klipper up. One of the BOOT pins is brought to a jumper, so I was able to flash the board using stm32flash and get the printer to work.

I'll post a config later.

64bittuning, 2020-05-11:
@ondra 
Whats your current status on this. Ive got a pair of machines running the ChiTu boards, and I have saved the stock firmware off them with STM32flash. One of the machines I got for free and it mechanically doesnt work worth a damn, so I dont mind beating it up.

Lampje83, 2020-06-16:
Shouldn't you set the crystal frequency to 8 MHz as the 12MHz crystal is used for the CH340 chip? Or don't you have 2 crystals on your board?

xC0000005, 2020-06-16:
I figured out what was wrong with the Chitu boards for Marlin, Klipper would need something similar. The long and short of it is that the bootloader hands off peripherals in a running state (sigh). To get Marlin and disk based update working, I do the following:
Disable all interrupts.
Reset all three CRs of USART1
Memset the whole SDIO to 0.
Set the end of ram (and therefore estack) to 60k.
Update the encryption script with the latest one from Marlin (I figured out a few bugs).
That plus the encryption scripts gives you disk-based update for Tronxy/Chitu/etc.
Or you can continue to use Klipper without the Chitu bootloader and it’ll work just fine.

palouf34, 2020-08-02:
I’m update my board  with one  marlin firmware with update.cdb that it’s ok But I want  now update  with klipper but do’t understand your solution. Can you  explain  wo script to use  for update  with klipper.

than’s for advance.