Klipper3d/klipper #2378: Moving out of bound during BED_MESH_CALIBRATE

madbuda, 2020-01-12:
Pulled the latest updates and noticed the changes in bed_mesh taking into account for probe offsets. I changed mesh_min and max to be sane 20,20 / 280,280 and my starting point is now normal, however it keeps trying to take the x out of bounds.

I've tried a few different iterations of mesh settings, min/max, probe count, etc.  with no differences

`[bltouch]
sensor_pin: ar18
control_pin: ar11
x_offset: -42
y_offset: -16
#z_offset: 1.72 # setting this higher means closer to the bed, lower means further away
pin_up_touch_mode_reports_triggered: False
samples: 2
`
`[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 20,20
mesh_max: 280,280
probe_count: 3,3
#fade_start: 1.0
#mesh_pps: 2,3
`
[klippy.log](https://github.com/KevinOConnor/klipper/files/4050295/klippy.log)

madbuda, 2020-01-12:
looks like I didn't do a m112 so I ran it again, here is the log

FYI, the generated points are out of the bounds I set in the printer.cfg (logically, I know how to resolve this now seeing what is being generated

Index |  Tool Adjusted  |   Probe
  0   | (62.0, 36.0)    | (20.0, 20.0)
  1   | (192.0, 36.0)   | (150.0, 20.0)
  2   | (322.0, 36.0)   | (280.0, 20.0)
  3   | (322.0, 166.0)  | (280.0, 150.0)
  4   | (192.0, 166.0)  | (150.0, 150.0)
  5   | (62.0, 166.0)   | (20.0, 150.0)
  6   | (62.0, 296.0)   | (20.0, 280.0)
  7   | (192.0, 296.0)  | (150.0, 280.0)
  8   | (322.0, 296.0)  | (280.0, 280.0)

[klippy.log](https://github.com/KevinOConnor/klipper/files/4050300/klippy.log)

Hywelmartin, 2020-01-12:
As it say [here](https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Mesh.md) the specified mesh_min/max is probe position, so the probe offset needs to be taken into account...

Arksine, 2020-01-12:
@madbuda It seems like you have it under control, I'm glad you found the output helpful.  As you can see, your your `mesh_max` is too close to the edge of the bed.  When you the probe's x_offset is added its resulting in points that move beyond `position_max` in the [stepper_x] section.

FWIW, in the future I will look into doing boundary checks before probing begins.

madbuda, 2020-01-12:
My probe offsets are negative... It seems to me that this should result in my start position being out of bounds not the other way.

Arksine, 2020-01-12:
No, although my statement above is unclear about that because I mention that the x_offset is "added", I did not mean that in a mathematical sense.

If you want to move the probe to a specific position, you must subtract the probe offsets from that position to determine the correct tool position.

madbuda, 2020-01-12:
I understand that, it just seems unintuitive to use probe position and not nozzle position

Arksine, 2020-01-12:
That is the way it previously worked.  It caused significant confusion because users thought that they were specifying the mesh location.  We reworked it to do just that, and changed the name of the options to make it clear that they are referring to the mesh location.

closed 2020-02-02 by KevinOConnor