Klipper3d/klipper #2379: Feature request: Ability to run gcode / macro before and after M84

vladbabii, 2020-01-12:
For my toolchanger i have quite a few macros that do various checks (to protect against mainly against my stupidity or slicer bugs). I would like to run some macros when steppers are disabled to set some internal variables and do some cleanup before and after running m84, with 2 different sets of commands if possible (or same commands run twice, i can identify via gcode variables which is which if needed)

The most important part is being able to run gcode before M84  to park the current tool, otherwise i risk dropping a 250+C tool somewhere inside the printer when M84 is run.

Thank you

KevinOConnor, 2020-01-13:
FYI, it should be possible to make a new macro (eg, DISABLE_MY_HEATER) that implements the required functionality and calls M84.  It is then possible to change the upstream callers to use the new macro instead of M84.

-Kevin

vladbabii, 2020-01-20:
The more i change the printer the less it will be compatible with slicers and other software. Maybe a more generic pre-gcode and post-gcode option to run gcode before/after any command would be a more general solution.

If I'm gonna implement one of these (been learning python recently), would it be considered to be included in klipper (if implemented well) or would this feature be rejected ? And if the place of this feature would not be in klipper main codebase, would you mind accepting a script to easily install modules from other github repositories ?

KevinOConnor, 2020-01-23:
One possible solution is to allow gcode_macro to override existing commands (perhaps with a rename of the existing command to some new name so that the existing command can be called from the gcode_macro itself).  This may also help with #2289.

I do my best to review and provide feedback to PRs that are submitted.  Alas, it's not really possible to say whether or not a PR would be merged without first seeing it.

-Kevin

vladbabii, 2020-01-24:
I was asking about if the idea would be accepted. Probably my implementation will not be up to par at first. I will try my hand at doing gcode replacement.

A different (crazy) idea - if gcode replacement is implemented then with a small change in the move command to allow multiple extruders at once it might be feasable mixing nozzle implementation .. more incentives for me to learn to add stuff to klipper.

I will try implementing this, but it will probably take a while.

KevinOConnor, 2020-02-13:
FYI, I added the ability for gcode_macro to override a builtin command (commit 9818055a).

-Kevin

vladbabii, 2020-02-13:
Just saw the commit. Thank you so much!

vladbabii, 2020-02-20:
Posted a G28 example replacement here https://github.com/KevinOConnor/klipper/issues/2537

This works great!

closed 2020-02-20 by vladbabii