Klipper3d/klipper #2380: Probe triggered prior to movement when mesh calibrating

metalinspired, 2020-01-12:
I tried setting probe_count to 6,6 and it starts probing as it should but stops half way.
Lowering the value to 5 and 4 didn't help as it always stops after third row.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4050558/klippy.log)

ld3300, 2020-01-12:
how long have you been using the probe.  Have you had issue with false triggers before?

metalinspired, 2020-01-12:
Since day one, never had any issues with it.
I doubt it's up to probe anyway since, when the message is triggered, probe is not being used and what I mean by that is: after the last point in third row (for X set to 6 and Y to more than 3 that would be 18th point) bed is moved on Y axis to where next probing should occur but it just stops there and triggers the message, without actually trying to probe.

Arksine, 2020-01-12:
I would recommend that you check your probe wiring.  Might be a lose connection, a pinched wire, or a broken wire .

metalinspired, 2020-01-12:
Indeed something is causing the probe to be triggered on certain area.
I'll have to investigate it.

closed 2020-01-12 by metalinspired