Klipper3d/klipper #2381: Unable to install - Setuptools dropped python 2.7 support.

LaZZer010, 2020-01-12:
Latest version of setuptools now requires Python 3.5 and has dropped support for 2.7:
https://github.com/pypa/setuptools/blob/master/CHANGES.rst

This results in an error when running ./klipper/scripts/install-octopi.sh script.
``Downloading https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl (583kB)
setuptools requires Python '>=3.5' but the running Python is 2.7.16
``

akatik, 2020-01-12:
Full setup script output on just installed Raspbian:

pi@raspberrypi:~ $ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
avr-libc is already the newest version (1:2.0.0+Atmel3.6.1-2).
binutils-arm-none-eabi is already the newest version (2.31.1-11+rpi1+11).
binutils-avr is already the newest version (2.26.20160125+Atmel3.6.1-4).
build-essential is already the newest version (12.6).
dfu-util is already the newest version (0.9-1).
gcc-arm-none-eabi is already the newest version (15:7-2018-q2-6).
gcc-avr is already the newest version (1:5.4.0+Atmel3.6.1-2).
libffi-dev is already the newest version (3.2.1-9).
libncurses-dev is already the newest version (6.1+20181013-2+deb10u2).
libnewlib-arm-none-eabi is already the newest version (3.1.0.20181231-1).
libusb-dev is already the newest version (2:0.1.12-32).
python-dev is already the newest version (2.7.16-1).
python-virtualenv is already the newest version (15.1.0+ds-2).
stm32flash is already the newest version (0.5-1).
virtualenv is already the newest version (15.1.0+ds-2).
avrdude is already the newest version (6.3-20171130+svn1429-2+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Also creating executable in /home/pi/klippy-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
setuptools requires Python '>=3.5' but the running Python is 2.7.16
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 1

LaZZer010, 2020-01-12:
Thank you akatik!

akatik, 2020-01-12:
Pull request to fix install-octopi.sh script: https://github.com/KevinOConnor/klipper/pull/2382/commits/94a51b4439c2b57e39dfacad357534ca025f45b2

LaZZer010, 2020-01-12:
That worked great akatik! Thank you for the quick fix.

closed 2020-01-12 by LaZZer010