Klipper3d/klipper #2383: Cannot install Klipper

laurencefourie, 2020-01-12:
I have a new printer but I am unable to install klipper. I get the error below:

pi@octopi:~ $ sudo virtualenv /home/octoprint/klippy-env
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/octoprint/klippy-env/bin/python2
Not overwriting existing python script /home/octoprint/klippy-env/bin/python (you must use /home/octoprint/klippy-env/bin/python2)
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/octoprint/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
setuptools requires Python '>=3.5' but the running Python is 2.7.16
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/octoprint/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 1

laurencefourie, 2020-01-12:
The install is incomplete therefor there is no log file.

Hywelmartin, 2020-01-12:
#2381

p3dstore, 2020-01-12:
I am having this exact same issue.

ronlh1, 2020-01-12:
Same issue with pi 4 octoprint v 1.3.12

pi@Octopi:~ $ ./klipper/scripts/install-octopi.sh

###### Running apt-get update...
[sudo] password for pi:
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Fetched 15.0 kB in 2s (9,446 B/s)
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
avr-libc is already the newest version (1:2.0.0+Atmel3.6.1-2).
binutils-arm-none-eabi is already the newest version (2.31.1-11+rpi1+11).
binutils-avr is already the newest version (2.26.20160125+Atmel3.6.1-4).
build-essential is already the newest version (12.6).
dfu-util is already the newest version (0.9-1).
gcc-arm-none-eabi is already the newest version (15:7-2018-q2-6).
gcc-avr is already the newest version (1:5.4.0+Atmel3.6.1-2).
libffi-dev is already the newest version (3.2.1-9).
libncurses-dev is already the newest version (6.1+20181013-2+deb10u2).
libnewlib-arm-none-eabi is already the newest version (3.1.0.20181231-1).
libusb-dev is already the newest version (2:0.1.12-32).
python-dev is already the newest version (2.7.16-1).
python-virtualenv is already the newest version (15.1.0+ds-2).
stm32flash is already the newest version (0.5-1).
virtualenv is already the newest version (15.1.0+ds-2).
avrdude is already the newest version (6.3-20171130+svn1429-2+rpt1).
The following packages were automatically installed and are no longer required:
  libva-drm2 libva-x11-2 libva2 libvdpau1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Also creating executable in /home/pi/klippy-env/bin/python
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
setuptools requires Python '>=3.5' but the running Python is 2.7.16
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 1

KevinOConnor, 2020-01-12:
The online python repositories broke the default raspberry pi "virtualenv" command.  I've added a workaround (commit 5fc92ff6).  Please grab the latest code and retry.  (See the FAQ for upgrade instructions.)

-Kevin

nsmela, 2020-01-12:
Hello everyone,

I also have this issue.  I installed today, got this issue.  Someone noted the master was updated, so here's a log of my attempt to use the new master. When that didn't work, I removed the klippy-env and tried again.

`------------------------------------------------------------------------------
This image comes without a desktop environment installed because it's not
required for running OctoPrint. If you want a desktop environment you can
install it via

    sudo /home/pi/scripts/install-desktop
------------------------------------------------------------------------------
OctoPrint version : 1.3.12
OctoPi version    : 0.17.0
------------------------------------------------------------------------------

pi@octopi:~ $ cd ~/klipper
pi@octopi:~/klipper $ git pull
remote: Enumerating objects: 3, done.
remote: Counting objects: 100% (3/3), done.
remote: Total 4 (delta 3), reused 3 (delta 3), pack-reused 1
Unpacking objects: 100% (4/4), done.
From https://github.com/KevinOConnor/klipper
   2f8ad5e6..5fc92ff6  master     -> origin/master
Updating 2f8ad5e6..5fc92ff6
Fast-forward
 scripts/install-octopi.sh | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)
pi@octopi:~/klipper $ ~/klipper/scripts/install-octopi.sh

###### Running apt-get update...
[sudo] password for pi:
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree
Reading state information... Done
avr-libc is already the newest version (1:2.0.0+Atmel3.6.1-2).
binutils-arm-none-eabi is already the newest version (2.31.1-11+rpi1+11).
binutils-avr is already the newest version (2.26.20160125+Atmel3.6.1-4).
build-essential is already the newest version (12.6).
dfu-util is already the newest version (0.9-1).
gcc-arm-none-eabi is already the newest version (15:7-2018-q2-6).
gcc-avr is already the newest version (1:5.4.0+Atmel3.6.1-2).
libffi-dev is already the newest version (3.2.1-9).
libncurses-dev is already the newest version (6.1+20181013-2+deb10u2).
libnewlib-arm-none-eabi is already the newest version (3.1.0.20181231-1).
libusb-dev is already the newest version (2:0.1.12-32).
python-dev is already the newest version (2.7.16-1).
python-virtualenv is already the newest version (15.1.0+ds-2).
stm32flash is already the newest version (0.5-1).
virtualenv is already the newest version (15.1.0+ds-2).
avrdude is already the newest version (6.3-20171130+svn1429-2+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 80 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Not overwriting existing python script /home/pi/klippy-env/bin/python (you must                                           use /home/pi/klippy-env/bin/python2)
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/pi/klippy-env/bin/python2 - setuptools pkg_                                          resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.                                          py3-none-any.whl
setuptools requires Python '>=3.5' but the running Python is 2.7.16
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
[... 66 lines cut ...]
stm32flash is already the newest version (0.5-1).
virtualenv is already the newest version (15.1.0+ds-2).
avrdude is already the newest version (6.3-20171130+svn1429-2+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 80 not upgraded.

###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Also creating executable in /home/pi/klippy-env/bin/python
Installing pip, wheel...done.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cffi==1.12.2
  Downloading https://files.pythonhosted.org/packages/64/7c/27367b38e6cc3e1f49f193deb761fe75cda9f95da37b67b422e62281fcac/cffi-1.12.2.tar.gz (453kB)
     |████████████████████████████████| 460kB 1.6MB/s
    ERROR: Command errored out with exit status 1:
     command: /home/pi/klippy-env/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-w6NkEQ/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-w6NkEQ/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-w6NkEQ/cffi/pip-egg-info
         cwd: /tmp/pip-install-w6NkEQ/cffi/
    Complete output (3 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ImportError: No module named setuptools
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
`

KevinOConnor, 2020-01-12:
Okay, thanks.  Hopefully it is fixed now (commit 730c7cfc).

If a previous install failed, it may be necessary to manually remove the previous "virtualenv" directory prior to rerunning the install-octopi.sh script:
```
rm -r ~/klippy-env/
```

-Kevin

austinvojta, 2020-01-12:
Disclaimer that I know very little what I'm doing, but have run into similar new Klipper install issues today. I pulled the latest Klipper as of 6 minutes ago and tried again to run install-octopi.sh, still got the same problem:

```
###### Updating python virtual environment...
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/pi/klippy-env/bin/python2
Not overwriting existing python script /home/pi/klippy-env/bin/python (you must use /home/pi/klippy-env/bin/python2)
Installing setuptools, pkg_resources, pip, wheel...
  Complete output from command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel:
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/
Collecting setuptools
  Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
setuptools requires Python '>=3.5' but the running Python is 2.7.16
----------------------------------------
...Installing setuptools, pkg_resources, pip, wheel...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
    download=download,
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/pi/klippy-env/bin/python2 - setuptools pkg_resources pip wheel failed with error code 1
```

glowtape, 2020-01-13:
I wonder if pip install needs the --upgrade parameter to actually do install the requested version of setuptools. Because the command in the log above doesn't mention anything about 44.0.

KevinOConnor, 2020-01-13:
Make sure you grab the latest code first - see the Klipper FAQ for upgrade instructions.

-Kevin

austinvojta, 2020-01-13:
> Make sure you grab the latest code first - see the Klipper FAQ for upgrade instructions.
> 
> -Kevin

Yep, on the latest version:
```pi@octopi:~/klipper $ git log
commit 730c7cfc207aec2199220601018c6fc2b9da0235 (HEAD -> master, origin/master, origin/HEAD)
Author: Kevin O'Connor <kevin@koconnor.net>
Date:   Sun Jan 12 18:16:43 2020 -0500

    scripts: Install "virtualenv setuptools==44.0.0" in install-octopi.sh

    Signed-off-by: Kevin O'Connor <kevin@koconnor.net>
```

nsmela, 2020-01-13:
I tried again, issue resolved for me!  Excellent work!

austinvojta, 2020-01-13:
> I tried again, issue resolved for me! Excellent work!

Did you start from scratch with formatted SD or just pull latest Klipper and retry?

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the original issue has been resolved.  (FYI, the original issue was the result of a change upstream to the Python package servers - which has been subsequently fixed.)

-Kevin

closed 2020-02-02 by KevinOConnor