Klipper3d/klipper #2384: Feature request: Menu based z-offset / probe adjustment

BoHoehne, 2020-01-12:
Please think about a menu based z-offset adjustment for probe offset adjustments thru the printers display.

I just tried to adjust the z-offset with the described method for a bl-touch sensor.
This process is really buggy. 
I had to enter "testz z=" (mind the pleonasm!) fifty or so times and not getting a good result because ether the sensor had an error and/or the "probe_calibrate" command had left the probing mode (without me noticing). 
Just to find out that I was off by one milimeter.

I am at a point where I seriously consider other projects. 

How about using a knob, that is already there.

Handycapped people are successfully repelled by such hurdles. 

Sorry, had to get this out of my system.

supermerill, 2020-01-15:
if it's a gui issue and if you're using octoprint, maybe you should post that on https://github.com/mmone/OctoprintKlipperPlugin

For what i know, klipper doesn't have a gui, the 'knob' have to be added inside octoprint or dwc2.

BoHoehne, 2020-01-15:
What i meant was the printers display unit.
In my case it is the reprap_discount_full_graphic controller with a rotary encoder aka. "knob".

So basically this would need a menu entry and sofware procedure to calibrate the probes z-offset.

arifhn, 2020-01-16:
I agree that the menu on LCD needs improvement.
But for now, you can create custom menu item for that command,

JohnEdwa, 2020-01-16:
The menu system is currently being rewritten [(#2344)](https://github.com/KevinOConnor/klipper/pull/2344), but I did create this exact system for my own use, [which you can find here.](https://gist.github.com/JohnEdwa/97689aae9b8179c5d3476d16a1dd3fc3) You need to change the BLtouch debug commands though, as they are for the new high-speed PR [(#2309)](https://github.com/KevinOConnor/klipper/pull/2309).

To actually use it you need to also copy one of the menu trees from [menu.cfg](https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/display/menu.cfg) to your printer config and add `__bltouch` to it.

Once both of those are finalized and merged, I'll create a proper PR for something like this.

closed 2020-01-19 by klipper-gitissuebot

dyang-7, 2021-09-02:
select testz,long click