Klipper3d/klipper #2385: Problem with clone bltouch during the probing phase

bigsubgit, 2020-01-13:
I'm having problems with probing with a clone bltouch, installed on a SKR V1.3. I tried a mix of parameters in configuration but without success.
The wrong value in this example is sent to terminal when the toolhead is up yet.
Could be a wrong configuration or other problems?

Recv: // probe at 0.003,19.999 is z=1.450000
Recv: // probe at 0.003,19.999 is z=1.450000
Recv: // probe at 90.005,19.999 is z=1.320000
Recv: // probe at 90.005,19.999 is z=9.090000
Recv: !! Probe samples exceed samples_tolerance
Recv: !! Probe samples exceed samples_tolerance
Recv: ok

[klippy.log](https://github.com/KevinOConnor/klipper/files/4053346/klippy.log)

KevinOConnor, 2020-01-14:
The probe is reporting inaccurate values.  It could be the configuration or it could be faulty hardware.  The best advice I have on configuring the bltouch is in the docs at: https://www.klipper3d.org/BLTouch.html

-Kevin

bigsubgit, 2020-01-14:
Thanks Kevin, are there possibilities that "bltouch" configuration section doesn't work well with clone bltouch?

darkporpoise, 2020-01-17:
@bigsubgit
I've got mine working fairly well with a 1st gen "Geetech 3D Touch" if you want some example settings

bigsubgit, 2020-01-17:
> @bigsubgit
> I've got mine working fairly well with a 1st gen "Geetech 3D Touch" if you want some example settings

Ohh with pleasure if you can.
Many thanks

darkporpoise, 2020-01-17:
Check the pin settings, yours will almost certainly be different! My 3D Touch is a bit rubbish but will get repeatable accurate readings if you give it enough tries. Hence the massive number of retries below. It means the process takes ages but doesn't need doing often.

`[bltouch]
sensor_pin: ^PB7
control_pin: PB5
pin_move_time: 1.500
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: False
x_offset: -44 
y_offset: -14
z_offset: 0
speed: 2
samples: 6
sample_retract_dist: 3.0
samples_result: median
samples_tolerance: 0.050
samples_tolerance_retries: 20`

bigsubgit, 2020-01-17:
So many tolerance retries! I used only 2 retries. I’ll trying asap. Thanks

darkporpoise, 2020-01-17:
I know 😐
I also spent several hours tweaking the grub screw on the 3D Touch and running PROBE_ACCURACY repeatedly to get that far.
Takes ages to get the mesh but I get great first layer results now.

I'm still using a separate Z Endstop for homing so I'm not relying on the probe otherwise.

bigsubgit, 2020-01-17:
> I know 😐
> I also spent several hours tweaking the grub screw on the 3D Touch and running PROBE_ACCURACY repeatedly to get that far.
> Takes ages to get the mesh but I get great first layer results now.
> 
> I'm still using a separate Z Endstop for homing so I'm not relying on the probe otherwise.

I'm using average as sample_result, less samples and I'm in doubt about how the screw on the top works. Despite the wrong values the probe accuracy is very good.

bigsubgit, 2020-01-17:
The second value is taken when z is up but the strange thing that in that position it don't do the probe. Is like there is a false contact anche the probe measures.
0:22:00.531: N119 G29
0:22:05.622: // probe at 1.804,19.999 is z=2.825000
0:22:11.218: // probe at 1.804,19.999 is z=11.465000
0:22:11.219: // Probe samples exceed tolerance. Retrying...
0:22:22.569: // probe at 1.804,19.999 is z=2.825000

darkporpoise, 2020-01-21:
Not sure if it's relevant but I read a guide to calibrating the 3D Touch using the grub screw in the top which talks about a point after which the probe won't reliably lower the sensing pin.

If your screw was too low then I wonder if you could be experiencing this. The solution might then be to turn the screw anti-clockwise (the guide a read suggested increments of 90° turns). If you kept track of the number of 90° turns you could easily revert the changes if it didn't help.

bigsubgit, 2020-01-21:
> Not sure if it's relevant but I read a guide to calibrating the 3D Touch using the grub screw in the top which talks about a point after which the probe won't reliably lower the sensing pin.
>
> If [...]

ok many thanks. I'll trying quickly.

KevinOConnor, 2020-02-14:
I'm going to close this ticket as it looks like the conversation has concluded.  We're not aware of any accuracy problems with the Klipper bltouch code, but we have seen many reports of poor performance with bltouch clone hardware.  (We've also seen many reports that tuning the hardware can improve performance.)

-Kevin

closed 2020-02-14 by KevinOConnor