Klipper3d/klipper #2386: Unable to run Second Extruder Motor

notshure, 2020-01-13:
Hello Kevin or who may read this,

I have made some progress with Klipper, have been able to add a macro to control my heated chamber. My printer has one some prints! One of those prints sports two V5 heat sinks. I added lines from other examples to printer.cfg. Buit an interface between the thermistor and the control board's (MKS GEN L V1.0) analogue input. Had success getting Klipper to control the second hot end, awesome. 

Now to drive the second filament feeder stepper motor. We hit the techno brick wall.
I added a TMC2130 to the mix and was able to communicate to it via SPI as per other explanations on Git hub and other sites. I have not seen that motor turn with this 5th axis. ie I know that the motor is fine as it works on another axis. I tried swapping out the driver for a stand alone type, in this case a 2130. Still no luck.

From the Octo Pi's terminal we try;

Send: T0
Recv: // Unknown command:"ACTIVATE_EXTRUDER"
Recv: // Unknown command:"SET_DUAL_CARRIAGE"
Recv: ok
[...]
Not queuing T1, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware

I tried doing the

cd ~/klipper/
make menuconfig

thing again but could not see any options for a second tool/extruder in case it "was reported as invalid by the firmware"

I am not sure if I understand all this but T0 and T1 (assume they stand for Tool 0 and Tool 1) allow for the offsets and I noticed T0 and T1 used in the Slic3rs G code.

Next (after homing) we tried to jog the axis above E (first extruder )

Send: G0 F25 F800
Recv: ok
[...]
Send: G0 G25 F800
Recv: ok
[...]
Send: G0 H25 F800
Recv: ok

The reply is ok but nothing turns.

I have spent a couple of days on this problem but now need to ask.

Where am I going wrong?

Thanks in advance and I appreciate the work. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/4053554/klippy.log)

glowtape, 2020-01-13:
Your Klipper build is too old and sits on a version that didn't yet implement the new toolhead handling via macros. You'll need to update Klipper as per FAQ.

Is your printer an IDEX? Asking because of the type of T0/T1 macros you defined, but talk about offsets instead. If it's a two nozzle toolhead, you'd probably have to look at sample-multi-extruder.cfg for setting it up.

notshure, 2020-01-13:
Thanks GlowTape. I will give that a try later when I get back from work. No not an IDEX but a rebuild of a Makerbot Cube. I replaced the controller with an MKS GEN L V1.0.  You are correct, it is a 2 nozzle tool head ie 2 rolls of filament, 2 feeder motors and 2 hot ends on the same carriage.   
I assumed I needed T0 and T1 defined because I tried to slice a model with slic3r set with 2 extruder's and read the G code. The G code was using T0 and T1 to change between the filaments. So there must be a way to get the slicer to do the offsets?
Should I be able to jog the second extruder stepper motor from the terminal as per other axis;

G0 X25 F500      ie move the x axis 25mm with a feed rate of 500
G0 Y25 F500      ie move the y axis 25mm with a feed rate of 500
G0 Z25 F500      ie move the z axis 25mm with a feed rate of 500 
G0 E25 F500      ie move the e axis 25mm with a feed rate of 500 

G0 F25 F500      ie move the f axis 25mm with a feed rate of 500?????????
Not sure if this is right. Need to update Klipper but also have go for now.

Thanks again GlowTape

glowtape, 2020-01-13:
Yes, you still need T0 and T1 to select each extruder. That's how it's been traditionally done, and is what the slicer keeps emitting.

Klipper did meanwhile remove these commands, requiring the user to re-implement it via macros, _if necessary_ (when you have more than one extruder). This would allow to handle things like offsets, IDEX, the Prusa MMU, Y-splitters or something entirely new completely on the firmware side.

I assume you referred to the configuration examples in the master branch on this Github, which are ahead of your current build, while using an older build.

_If you are gonna update your Klipper build_, you will have to keep the macros, but define them similar as in the sample-multi-extruder.cfg I referred to. You'll likely need it like this:

```
[gcode_macro T0]
gcode:
    SET_GCODE_OFFSET X=0 Y=0
    ACTIVATE_EXTRUDER EXTRUDER=<extruder1name>

[gcode_macro T1]
gcode:
    SET_GCODE_OFFSET X=<offset> Y=<offset>
    ACTIVATE_EXTRUDER EXTRUDER=<extruder2name>
```
Insert the proper offsets and extruder names as per configuration. If you're using these macros, you can keep all offsets in the slicer to 0.

_If you are **not** gonna update your Klipper build_, you can remove the G-Code macros from your configuration. You will have to define the toolhead offsets in the respective configuration options in your slicer.

Example PrusaSlicer/slic3r:
[image]

Example Simplify3D:
[image]

glowtape, 2020-01-13:
Casually glancing over your configuration embedded in klippy.log, you're a) only configuring the TMC2130 for the second extruder and not the first one, and b) 0.1A run current at 64 microsteps, that's not a lot.

glowtape, 2020-01-13:
Sorry I didn't spot this one earlier:

> Not queuing T1, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware

To get rid of this one, you need to set the number of hotends to 2 inside the printer profile in Octoprint.

notshure, 2020-01-14:
Hi GlowTape,

Your advice was extremely helpful!  I updated Klipper which then cleared the "Recv: // Unknown command:"ACTIVATE_EXTRUDER" problem,  I assume the new version recognises this command. 
Your observation to "set the number of hotends to 2 inside the printer profile in Octoprint" was a gem! That got over the next hurdle and we finally got the motor to turn!!!!
I set the run and hold currents low on purpose. These little drivers get hot, so I like to start low.
I have ordered a 5160 (the only silent driver I could find with external MOSFETs that is capable of switching more than just a couple of amps).
I am using the 2130 (all I have at the moment) till the 5160 arrives.
I do have a couple more questions which may be answered else where, so just point me there if you know.
When using the  "DUMP_TMC stepper=extruder1" command, the returned values are a little cryptic 
ie IHOLD_IRUN: 00080602 IHOLD=2 IRUN=6 IHOLDDELAY=8
the IRUN went from 2 to 6  when setting the run current from 0.1 to 0.2 respectively 
ie
Set in Config          DUMP shows
0.1                          2
0.2                          6

Is there a formula or table to look up the meaning of these values and other parameters such as pwm_freq=1 PWM_AMPL=128 PWM_GRAD=4 etc??

My motor is NOISY whilst stopped/holding/braking. Is there a way of increasing the PWM frequency??  (I admit I have not hunted around for the answer to this one and understand that it may be a dumb question)

Once again THANK YOU very much. I would have never thought to look in OctoPi's settings.

glowtape, 2020-01-14:
I don't have particular experience in parsing these TMC config dumps. I do use TMC2209s and had TMC2130s in the past, but I just set some values and they worked, never had any reason for deep dives.

That said, the higher the microstepping, the less torque the steppers will have. That includes holding torque. That's why I was suspicious about the rather high microstepping of 1/64th in combination with the very low run current of 0.1A. If it's noisy while holding, in my opinion, it has trouble holding the position. If the printer is 12V instead of 24V, this would get even worse.

I'd get some rudimentary cooling and increase the current. Try to get towards 0.8A-1A. As far as heat goes, if the driver gets to 50-70°C, that's not really that problematic. I think the Tj for the TMCs is like 120°C, but I'd try to avoid getting that high.

Another thing you could try is disabling Stealthchop. It can be problematic. At lower currents, the Spreadcycle mode has mode torque. Add `stealthchop_threshold: 0` for that in the tmc2130 section.

Personally, I run a 4:1 geared extruder that's still 0.8A/24V at 1/16th microstepping. I wouldn't dream running it at 0.1A.

KevinOConnor, 2020-01-14:
> When using the "DUMP_TMC stepper=extruder1" command, the returned values are a little cryptic

The dump corresponds to the information in the trinamic specification for the drivers.  Do a web search for "trinamic tmc2130 datasheet".

-Kevin

notshure, 2020-01-16:
Hi GlowTape and Kevin,

I have tried adjusting the 2130's "run_current", "hold_current", "stealthchop_threshold" and "microsteps" from the "printer.cfg" file. After each change I save the file then go over to the terminal in octo pi and do a "RESTART_FIRMWARE". Once this is done I like to confirm the changes by running "DUMP_TMC stepper=extruder1". When I said cryptic, I meant that some values like the run current and hold current do NOT matching what is in the config file.
For example I set the run current to 0.4 and the hold current to 0.2 but the returned values from the DUMP command are 12 and 6 respectively. The numbers returned from the DUMP command do increase as the config values are increased. Changing these parameters like this is very cool!
I did spend 5 minutes looking at the 2130 data sheet - WOW, I think we are going to need a little more than 5 minutes with this glossy number. 

But every time I ran it, it was noisy. I know that the parameters are updating because the motor (on the desk with a masking tape flag rotating ie unloaded) will turn more or less time as I change the micro stepping. Tried disabling Spreadcycle. It is noisy running and noisy whilst holding.  
I also changed the PWM frequency using "SET_TMC_FIELD STEPPER=extruder1 FIELD=pwm_freq VALUE=2" (assume f PWM =2/512 f CLK (from pg 46 of the data sheet)) but no luck.

"Also, come to think of it, what driver are you running on your first extruder?"
The first 4 drivers are TMC2100 in stand alone mode, running on 24 volts. These are are amazing in terms of sound ie they are very mouse like quiet. Only problem is they do run hot. I have this 2130 and decided to use it for the second extruder for now (not because it boasts extra switching current but for the SPI bus as I have ordered a 5160 which does boast higher current switching. So trying to understand how to setup etc ) 
I will continue to play, most likely over the weekend.
I will have more of a read.
Thanks again to both of you.
Oh Kevin, that chamber heater control that you helped me with a while ago has been working well!
Hope to apply the lessons to a water cooled extruder that I plan to add in a few months.
Thanks again

notshure, 2020-01-16:
Found the problem, we now have SILENCE!  I swapped the stepper motor and the noise has gone. Silent while running and stopped. 
Not sure what the difference is between the motors but the second one is the cheese compared to the first one being chalk.

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2020-02-02 by KevinOConnor