Klipper3d/klipper #2387: MKS SGEN L + MKS Mini 12864, pin P0.17 for [display] cs_pin

akatik, 2020-01-13:
Board MKS SGEN L, pins: https://github.com/makerbase-mks/SGEN_L/blob/master/Hardware/MKS%20SGEN_L%20V1.0_001/MKS%20SGEN_L%20V1.0_001%20PIN.pdf

EXP1_6 is P0.17

Display MKS Mini 12864, pins: https://github.com/makerbase-mks/MKS-SBASE/issues/101#issuecomment-552074421

EXP1_6 is DOGLCD_CS

Seems display cs pin is P0.17.

Also, in Marlin config for MKS SGEN_L: https://github.com/makerbase-mks/SGEN_L/blob/master/Firmware/Marlin-bugfix-2.0.x/Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h

#if ENABLED(MKS_MINI_12864)
          #define DOGLCD_CS  P0_17

Also, in Smoothieware config for MKS SGEN_L: https://github.com/makerbase-mks/SGEN_L/blob/master/Firmware/Smoothieware/MKS%20MINI12864%20and%20MKS%20LCD12864A/config.txt

# MKS LCD12864/MINI12864控制面板设置
panel.spi_cs_pin                            0.17

I write to printer.cfg:

[display]
cs_pin: P0.17

And receive "error: pin P0.17 is reserved for ssp0"

How to run this display on this board with klipper?

[klippy.log](https://github.com/KevinOConnor/klipper/files/4053995/klippy.log)

Hywelmartin, 2020-01-13:
have you tried 
```
[display]
lcd_type: uc1701
cs_pin: EXP1_6
a0_pin: EXP1_7
contrast: 40
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

sometimes.. often.. the display connectors are rotated.. https://reprap.org/wiki/MKS_MINI_12864#Physical_Interface

akatik, 2020-01-13:
With "cs_pin: EXP1_6" (alias EXP1_6=P0.17) , I receive "error: pin EXP1_6 is reserved for ssp0". 

Display connection is not rotated. I have try Marlin and Smoothieware, both with display cs pin P0.17 (and other settings as in makerbase GitHub), and display works.

KevinOConnor, 2020-01-13:
If the board is really wired up such that P0.17 is used for cs_pin, then you'll need to use software spi and explicitly list the pins used for miso, mosi, and sclk.  (See the comments in config/sample-lcd.cfg for further details.)

-Kevin

akatik, 2020-01-14:
With those lines in config:

```
cs_pin = P0.17
spi_software_miso_pin = P0.8
spi_software_mosi_pin = P0.9
spi_software_sclk_pin = P0.7
```

I receive "error: MCU 'mcu' error during config: oids already allocated". Attached log software-spi-klippy.log

Display runs with hardware SPI ssp1. Code in PR #2393 Log in spi1-klippy.log
[software-spi-klippy.log](https://github.com/KevinOConnor/klipper/files/4060499/software-spi-klippy.log)
[spi1-klippy.log](https://github.com/KevinOConnor/klipper/files/4060500/spi1-klippy.log)

KevinOConnor, 2020-01-14:
The software spi error looks like something odd happened during configuration - you should be able to clear that error with a FIRMWARE_RESTART command.  (If not, could you post a log that includes the FIRMWARE_RESTART.)

Clearly using hardware ssp1 is the preferred solution, however.

Thanks,
-Kevin

akatik, 2020-01-14:
Something odd happened. I rebuild everything and now it works with software SPI. Thank you.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4061541/klippy.log)

KevinOConnor, 2020-01-15:
Separately, it would be good if we could add a config/generic-mks-sgen-l.cfg example config file.

-Kevin

KevinOConnor, 2020-02-02:
I'm going to close this as it looks like the original issue was resolved.

-Kevin

closed 2020-02-02 by KevinOConnor

DIYlover, 2021-03-28:
@akatik Hello! I too MKS SGen L user.  I’ve got difficult to flash Klipper, could u mind share me the procedure how you can flash Klipper firmware to this board?

Zener28, 2021-09-09:
Hi @akatik , I have a trouble to start the Mini LCD 12864 with a MKS SGENL V1.0, could you share with me your firmware to enable the lCD?

DIYlover, 2021-09-10:
> Hi @akatik , I have a trouble to start the Mini LCD 12864 with a MKS SGENL V1.0, could you share with me your firmware to enable the lCD?

Hello @Zener28 Not sure my config is good for your reference.  My BTT Mini LCD 12864 is actually connected to no brand SKR1.3 which working along with MKS SGENL V1.0 as [mcu z] for handling 4 Z-stepper motors in Voron 2.4.  Below is display section of my printer configuration for your reference.

#####################################################################
# 	Displays
#   by BTT/Fysetc Mini 12864Panel v2.1 (with neopixel backlight leds)
######################################################################

[display]
lcd_type: uc1701
cs_pin: EXP1_3
a0_pin: EXP1_4
rst_pin: EXP1_5
contrast: 63
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
## Some micro-controller boards may require an spi bus to be specified:
#spi_bus: spi
## Alternatively, some micro-controller boards may work with software spi:
#spi_software_miso_pin: EXP2_1
#spi_software_mosi_pin: EXP2_6
#spi_software_sclk_pin: EXP2_2

#[output_pin beeper]
#pin: EXP1_1

[neopixel fysetc_mini12864]
pin: EXP1_6
chain_count: 3
color_order: RGB
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 1.0

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"