Klipper3d/klipper #2388: Z-axis steps are not applied no matter what value

BoHoehne, 2020-01-13:
Hello,

I am having a strange problem.
I can not calibrate the z-axis, because the step_distance value is not applied.

I tried the following:

Edit the printer.cfg
Restart klipper service.
Issued a firmware_restart command.
(This did not work, still the old setting)

Then I powercycled the printer.
Issued a firmware_restart.
Measured again, no change.
(This did not work, still the old setting)

The z-axis alway moves 19mm instead of 100mm.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/4054091/klippy.log.zip)

The z-axis a uart controlled tmc2208:

Send: dump_tmc stepper=stepper_z
Recv: // ========== Write-only registers ==========
Recv: // IHOLD_IRUN: 00081913 IHOLD=19 IRUN=25 IHOLDDELAY=8
Recv: // TPWMTHRS:   00000139 TPWMTHRS=313
Recv: // TPOWERDOWN: 00000014 TPOWERDOWN=20
Recv: // ========== Queried registers ==========
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000001e IFCNT=30
Recv: // OTP_READ:   00000000
Recv: // IOIN@TMC220x: 20000140 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
Recv: // FACTORY_CONF: 00000000
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000003c8 MSCNT=968
Recv: // MSCURACT:   00e901ad CUR_A=-83 CUR_B=233
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0130000 CS_ACTUAL=19 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00000019 PWM_SCALE_SUM=25
Recv: // PWM_AUTO:   000e0028 PWM_OFS_AUTO=40 PWM_GRAD_AUTO=14

All other steppers work normally.
Any hints are welcome.

glowtape, 2020-01-13:
Does this log file contain multiple restarts where you changed the step_distance? Because every time Klipper restarts and dumps the configuration into the log file, step_distance is always .0125. Did you edit the right printer.cfg in the right location (home directory)?

BoHoehne, 2020-01-13:
OK, that was a good hint.

Apparently, along the way, klipper service decided to generate a new config file.
I didn't see that one coming.

Unless someone implements a foolproof configuration management solution, this behavior is a recipe for catastrophe.

Thank you.

puneit-hassio, 2020-01-13:
This happens if you edit config file via octoprint. Do it directly from terminal or via filezilla.

BoHoehne, 2020-01-13:
I did edit it directly via the ssh terminal.

As an old sysadmin i created a symlink to the original config file. I didn't expect klipper to throw that away and create a non symlinked copy.

Because of this i had to redo the z-axis offset calibration. 
It failed in most cases because it suddenly doesn't accept the testz z=... command as "unknown command" or the "accept" and "save_config" is not reflected in any config file. 

This is really annoying.

glowtape, 2020-01-13:
Oh the symlink thing. Klipper creates a copy of the file when doing something like SAVE_CONFIG, then renames everything and deletes the old copy.

BoHoehne, 2020-01-15:
This should probably documented better. 

I'm closing this for now.

BoHoehne, 2020-01-15:
Now.

closed 2020-01-15 by BoHoehne