Klipper3d/klipper #2389: BUGFIX: config/generic-bigtreetech-skr-pro.cfg

talfari, 2020-01-13:
Hi,

i'm setting up my new printer with the a combination of klipper and a Bigtreetech SKR pro v1.1. Its working very well as far as I'm going builden the machine. So THANK YOU for all the work you are puuting in this project.

Setting up the thermistors I found, that the default pin configuration of the configuration file is wrong. 

The Sensor header are permuted. The assigned pins should be:

sensor_pin: PF3 # T0 Header  <<==== HEATER_BED, not E0!
sensor_pin: PF4 # T1 Header  <<==== E0, not E1 
sensor_pin: PF5 # T2 Header  <<==== E1 ...
sensor_pin: PF6 # T3 Header  <<==== E2 ...

Maybe you could update the file.

Regards

Andreas

talfari, 2020-01-13:
... of course you could wire up the things the other way around, but that's the intended/documented way by Bigtreetech.

Hywelmartin, 2020-01-13:
[image]
[image]

talfari, 2020-01-13:
Exactly. In the file, the Bed is set to sense on port pf6, but should be pf3.

Supplied file:
T0 -> E0 : pf3
T1 -> E1: pf4
T2 -> E2: pf5
T3 -> Bed: pf6,

but it should be:
T0 -> Bed : pf3
T1 -> E0: pf4
T2 -> E1: pf5
T3 -> E2: pf6

Hywelmartin, 2020-01-15:
@talfari fixed ???

closed 2020-01-15 by talfari