Klipper3d/klipper #2394: [Feature request]Fan off below modification

GarrethX, 2020-01-14:
Hello, 

I'm looking for the reason behind function off_below in fan section. I know that fan can stall, it happened to me, but why do we turn off the fan completely and not set it for a minimal value?
For example if we have off below 50%, and slicer requests something between 1-49%, why not have a parameter which would turn the fan to minimum of 50%? 
We could have actually both parameters, minimum_fan_speed and off_below, for example off below 10%, and minimum fan speed 50%. 

What do you think about such functionality?

KevinOConnor, 2020-01-15:
FYI, this was discussed a bit in PR #1897 and PR #2165 .

-Kevin

GarrethX, 2020-01-15:
Hey,

Thanks Kevin for finding these PR, first one I found myself, second one not.  
First PR mentioned is about off_below. It was suggested with a different name, but eventually became off_below function which I think works correctly. 
Second one sounds a bit like overcomplication of the thing because of scaling, I don't think that's necessary. 

In Simplify3D I can set up to increase fan speed for layers below some time. As a result sometimes I receive M106 values which are below the threshold of off_below value (which I need as my fans apparently like to stall, I burned on of them last week because of that issue). Unfortunately it is easier to make a change in the klipper than to expect slicer modification:)

I was thinking about adding few simple lines in the fan code:

`        self.min_power = config.getfloat(
            'min_power', 0., above=0., below=self.max_power, maxval=1.)`

`        if value < self.off_below:
            value = 0.
        elif value < self.min_power:
            value = self.min_power`

Something like above - I don't think it turns klipper fan functionality upside down, but maybe to few people would be useful.

KevinOConnor, 2020-01-23:
You should find a "minimum fan speed" setting in your slicer.  For correct operation, this really needs to be configured there (and not in the firmware).  The off_below setting is primarily intended as a safety feature (that is, to prevent the fan burning out even if the slicer is mis-configured).

-Kevin

GarrethX, 2020-02-01:
Hey,

Since some time I switched slicer to Cura and fan settings are far more advanced but also manageable there, and you're right that it should be properly set during slicing. However in s3d there is no minimal fan speed setting and when I played with option to have fan speed set according to layer time I managed to receive fan requests below my off_below speed and failed a print.

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like a solution was found for the original issue (set the fan speed in the slicer).  As for Klipper working around a deficiency in s3d - FWIW, I don't see much value in doing that as it will never be a good solution - this is something that really needs to be addressed in the slicer.

-Kevin

closed 2020-02-14 by KevinOConnor