Klipper3d/klipper #2395: MCU crash on start print

manoukianv, 2020-01-14:
Hi,

i have a crash on the start printing.
What expected : home, execute "START_GCODE", wait temperature, move head down, start to print.
What i have : home, execute "START_GCODE", wait temperature, move head down, MCU/Klipper issue.

I Have the issue with the screen startup (glcd) , and when i use octoprint.
I have TMC2209 in sensorless on X/Y, and with a probe on Z.
[klippy.log](https://github.com/KevinOConnor/klipper/files/4061609/klippy.log)
[Y holder 10mm.gcode.zip](https://github.com/KevinOConnor/klipper/files/4061616/Y.holder.10mm.gcode.zip)

Thanks for you help
Vincent

KevinOConnor, 2020-01-23:
It doesn't look like you are using code from this repository.  If you're using code from @dmbutyugin's repo, then note that you should upgrade to the latest code as there were several fixes there.

-Kevin

KevinOConnor, 2020-02-02:
I'm going to close this as there isn't much we can do with reports from external code.

-Kevin

closed 2020-02-02 by KevinOConnor

dmbutyugin, 2020-02-05:
@manoukianv I looked at this case. As Kevin suggested, you need to sync to the latest code, there were some fixes in the mainline related to scheduling I believe. Also, please try the same code with the main Klipper branch and see if the issue is still reproducible, or if it only happens on S-Curve branch. If it reproduces only on S-Curve branch, feel free to continue the conversation in the issue #57, which is related to S-Curves.

FWIW, I noticed an abnormality in the logs that around the crash there are many (>1000) moves queued, which suggests many tiny moves and high move rate. Are you printing from the virtual_sdcard? If not, there is a chance that Octoprint cannot keep up with the move rate, which may exacerbate the issue.