Klipper3d/klipper #2398: MKS TFT 2.4

jcbarrado, 2020-01-15:
I have a MKS TFT 2.4 with a MKS Ramps 1.4 and I would like to use the TFT with klipper. Could you help me?

Coffee0297, 2020-01-16:
TFT is not supported by klipper at this time.

-Tonny

softkannan, 2020-01-16:
Can we provide generic serial (UART) module so that Touch display support can be done in python easily? 
I know each TFT has custom protocols. but once we have generic UART module then others can easily write serial command handling module in python and we can use stock display.
I am talking like similar to “tmcuart.c” but just writeline, onrecevline, something along that line of commands and sample python implementation.

timmit99, 2020-01-16:
@jcbarrado do you have any way to see what the serial output of the TFT is when you hit certain buttons? If we can make a map of what commands it is sending then what @softkannan is proposing would work. just interpret the serial commands into klipper commands the host can use. Either that or just make a serial pass through and make G-Code macros for things.

I can try hooking my TFT up to a serial sniffer and see what i get for outputs.

KevinOConnor, 2020-01-21:
This looks like a dup of #2231 .

-Kevin

DeviousPenguin, 2020-01-30:
> I can try hooking my TFT up to a serial sniffer and see what i get for outputs.

You may be able to use a USB to Serial adaptor  up to the RPi in order to get commands from the TFT screen.

KevinOConnor, 2020-02-02:
I'm going to close this in favour of the existing #2231 .

-Kevin

closed 2020-02-02 by KevinOConnor