Klipper3d/klipper #2402: SKR 1.3 bltouch internal G28 error when homing

khessels, 2020-01-17:
[klippy.log](https://github.com/KevinOConnor/klipper/files/4075120/klippy.log)

Hei Kevin, i have been struggling with this for the last week, but whatever i do i get this message.
i have verified that the bltouch is working with the pin down and up command.
I have installed the latest skr firmware and that did not solve anything.
2208 on the x, y and z axis, 8825 on the extruder. trying to use the bltouch as endstop for the z.
The log indicates an fp error divide by zero with max_steps, but o have no idea where to set that...
i could not find it in the sample scripts\docs... and not yet very familiar with python debugging at the moment...

ld3300, 2020-01-18:
Hi khessels,

You do have a couple of oddities in your config.  Your BLtouch offset is unlikely to be 0.  That would mean the nozzle is touching the bed when the probe triggers. set this to something like 1 then do a probe calibrate to get the right value.

You should remove the homing_retract_dist from your z axis as it is meaningless with the bltouch config, and it is possible it is doing something unintentional.

khessels, 2020-01-18:
made the changes as you suggested.. but alas, no dice.... if i leave the bltouch out it does home (but i have no z endstop)

ld3300, 2020-01-18:
Do your X and Y axis home?  It looks like you have told klipper that the endstop sensors are at position 0, but you have homing_positive_direction set to true.

khessels, 2020-01-18:
I removed the blt and added a regular endstop (back to basics right).
it looks like i did not have the "position_endstop" specified in my z stepper and that may have caused the problem. Closing the issue.

closed 2020-01-18 by khessels