Klipper3d/klipper #2404: Question- Is it necessary to update MCU firmware?

scarecrow2k6, 2020-01-17:
I am running Klipper on top of the Smoothie firmware. Is it necessary to update the Smoothie firmware if I want any new features? Or is it that Klipper overrides and becomes the primary fw and only uses the MCU for the hardware? I hope my question made sense, I'm new to this.

timmit99, 2020-01-17:
Klipper *is* the firmware. When you flash the MCU's with the klipper.bin from the `make menuconfig` and `make flash` commands it erases whatever was on the MCU and puts the klipper "slave" code on it. there would be no smoothieware left so you only have to update the Klipper code and possibly reflash if the code requires an MCU update.

This make sense?

scarecrow2k6, 2020-01-17:
> Klipper _is_ the firmware. When you flash the MCU's with the klipper.bin from the `make menuconfig` and `make flash` commands it erases whatever was on the MCU and puts the klipper "slave" code on [...]

Yes absolutely, for some reason I thought Klipper ran on top of the original fw. Didn't know it actually replaced it. The reason I say that because I have been using a Raspberry Pi + Klipper, I thought Klipper ran on RPi on top of Smoothie (Smoothie being dormant). Thanks for your answer!

timmit99, 2020-01-17:
Youa re very welcome and welcome to Klipper. Having a single file to edit your settings and not having to flash EVERY time you want to change a minuscule setting is so worth it!

GarrethX, 2020-01-22:
I'll add my 5 cents to that question, and ask - does Klipper report in Octoprint that the MCU firmware has to be updated?

KevinOConnor, 2020-01-23:
This is answered in the FAQ - https://www.klipper3d.org/FAQ.html#how-do-i-upgrade-to-the-latest-software .  In short, yes, the software should inform the user if a flash is necessary.

-Kevin

closed 2020-02-02 by KevinOConnor