Klipper3d/klipper #2406: Replicape Cant Connect Using dual Extrudes.  enable_pin: replicape:stepper_h_enable

vvvteddybearvvv, 2020-01-19:
I am unable to get octoprint to connect to due to a config file issue with the Replicate. i have a dual extruder setup using the following config snippit

[extruder1]
step_pin: P8_11
dir_pin: P8_16
enable_pin: replicape:stepper_h_enable
step_distance: 0.038227
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: replicape:power_h
sensor_type: ATC Semitec 104GT-2
sensor_pin: host:analog5

Octo print is giving me the following error. 

Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Unknown replicape pin stepper_h_enable

i am unable to get the log files in the directions due to octoprint being offline

closed 2020-01-19 by vvvteddybearvvv

BlackStump, 2020-01-19:
The log is not obtained through Octoprint. please re-read the instructions.
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

An "scp" and/or "sftp" utility is needed to acquire this log file. The "scp" utility comes standard with Linux and MacOS desktops. There are freely available scp utilities for other desktops (eg, WinSCP).
Use the scp utility to copy the /tmp/klippy.log file from the Raspberry Pi to your desktop.

BlackStump, 2020-01-19:
I enabled H to see if there is a problem this is what I used
[extruder1]
step_pin: P8_11
dir_pin: P8_16
enable_pin: replicape:stepper_h_enable
step_distance: 0.0024096385542169
nozzle_diameter: 0.500
filament_diameter: 1.750
heater_pin: replicape:power_h
sensor_type: NTC 100K beta 3950
sensor_pin: host:analog5
control: pid
pid_kp: 20.470
pid_ki: 1.026
pid_kd: 102.099
min_temp: 0
max_temp: 270

vvvteddybearvvv, 2020-01-19:
I was able to get it to work. I forgot to set the current and that caused
it not to enable. Sorry for the fire drill