Klipper3d/klipper #2410: Fysetc Cheetha 1.1b Printer not ready

Magoo0876, 2020-01-21:
I recently bought a new version of the Cheetha board 1.1b. Unfortunately there is no schematics for that, only 1.1 and 1.2 is available. I've found out that it is a mixture of both. 

So I flashed the thing like in the config described and changed the printer.cfg.
Everything looked good but the host can not connect to the board. 

What realy is strange, if you connect the usb cable, the onboard 5V are dropping to 4.4V. With the marlin sw you can see hollow letters on the display. Its fine again if you disconnect. To mitigate that I changed the USB fuse to a jumper to disconnect the 5V from the cabel. The display is fine now, but still printer not ready. 

I did all the cheap things like reflashing and id checking.
So if you have any idea please be so kind and help. 

I've bought that board to compare with my troublemaker SKR mini E3, and now another nice board from china. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/4088443/klippy.log)

Magoo0876, 2020-01-27:
So I tried to connect Cura via Marlin and Windows, that worked so far. New clean flash and Klipper + Raspberry nothing. 
Log:
Timeout on serial connect
Traceback (most recent call last):

  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response_

                  [ ] Enable extra low-level configuration options                                    │ │  
                      Micro-controller Architecture (STMicroelectronics STM32)  --->                  │ │  
                      Processor model (STM32F103)  --->                                               │ │  
                      Bootloader offset (No bootloader)  --->                                         │ │  
                 [ ] Use USB for communication (instead of serial)      

Maybe I'm stupied but I think the config is ok.

KevinOConnor, 2020-01-28:
Alas, without a schematic for the board, I"m not sure there's much we can do.  You'll need to know the micro-controller type, how it's wired for external communication, and what type of bootloader it is using.  With that info, it should be possible to get Klipper running on it.

-Kevin

KevinOConnor, 2020-02-29:
Unfortunately, there was an error in the stm32f1 code just recently fixed (commit 31b2c3ea).  Best to upgrade to the latest code and retry with that.

-Kevin

Magoo0876, 2020-03-14:
Hi Kevin,

I've allready tried what you suggsted, but no change so far. I could arrange to get the Altium files for the board from the designer, but it look very simmilar to the allready known boards. The new log is also attached.

hth thanks

 

[Cheetah V1.1B.zip](https://github.com/KevinOConnor/klipper/files/4333318/Cheetah.V1.1B.zip)
[klippy.log](https://github.com/KevinOConnor/klipper/files/4333321/klippy.log)

Magoo0876, 2020-03-14:
So good news! I solved my problem by updating the Raspberry firmware.
So simply type at the commandline of your pi:

sudo rpi-update 

Reboot afterwards, and pronblem should be fixed.
If not do a software update too:

sudo apt full-upgrade

-Magoo

closed 2020-03-14 by Magoo0876