Klipper3d/klipper #2411: Internal error on command:"G1"

gjnijenhuis, 2020-01-21:
During printing I received the error 'Internal error on command:"G1"' which caused the print to fail.
I am on the master branch and updated last week. I'm on version v0.8.0-231-g3fb0b7e4

Looking in the klippy.log the following section shows more about the issue:

`stepcompress o=4 i=0 c=4 a=547: Point 4: 3282 not in 2880:3280
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 233, in _process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 480, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 166, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 409, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 180, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 169, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 327, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 280, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 167, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress`

It looks like it is related to #2225 but the error message from the stepcompress is different (`Point 4: 3282 not in 2880:3280` vs `Invalid sequence`

[klippy (1).log](https://github.com/KevinOConnor/klipper/files/4090102/klippy.1.log)

KevinOConnor, 2020-01-23:
It looks like you've hit a bug in Klipper.  Unfortunately, I'm unable to reproduce the problem locally from the information in the log file.  Could you attach the original gcode file you were printing at the time?

-Kevin

gjnijenhuis, 2020-01-24:
Sure, see attached.
[65mm_roll x2 85mm_roll x2 main_body shaft_158mm x2.zip](https://github.com/KevinOConnor/klipper/files/4108722/65mm_roll.x2.85mm_roll.x2.main_body.shaft_158mm.x2.zip)

KevinOConnor, 2020-01-25:
I think this should be fixed now (commit 5ca132ea).

-Kevin

closed 2020-02-14 by KevinOConnor