Klipper3d/klipper #2412: Resume after a pause if not always successful

boardthatpowder, 2020-01-22:
Using my `PAUSE_MACRO` followed by `RESUME_MACRO` is not always successful.  Sometimes after resuming it causes unexpected behavior such as:
- moving Z below the print, crashing into it (what the attached log represents)
- moving Z way above the print, extruding into thin air
- extruding lines is think I would in random directions I would not believe possible

This means I can't risk running out of filament at any time, and am unable to pause and resume a print with any confidence.  

```
[gcode_macro PAUSE_MACRO]
default_parameter_x = 20
default_parameter_y = 280
default_parameter_z = 50
gcode = 
	SAVE_GCODE_STATE NAME=PAUSE_state
	PAUSE
	RESPOND TYPE=command MSG=action:paused
	PARK_MACRO X={X} Y={Y} Z={Z}
	SET_IDLE_TIMEOUT TIMEOUT=3600

[gcode_macro PARK_MACRO]
default_parameter_x = 20
default_parameter_y = 250
default_parameter_z = 30
gcode = 
	;M117 PARKING..
	SAVE_GCODE_STATE NAME=parking
	G91
	G1 E-2 F1000
	G1 Z{Z}
	G90
	G1 X{X} Y{Y} F3000
	RESTORE_GCODE_STATE name=parking

[gcode_macro RESUME_MACRO]
gcode = 
	;M117 Resuming...
	SET_IDLE_TIMEOUT TIMEOUT=600
	RESTORE_GCODE_STATE NAME=PAUSE_state
	RESUME
	RESPOND TYPE=command MSG=action:resumed
```

Any insights will be greatly appreciated.

[klippy.log](https://github.com/KevinOConnor/klipper/files/4095060/klippy.log)

ld3300, 2020-01-22:
did you try `RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1`

The move parameter tells it to also restore to the previously saved XYZ coordinates

Also, are you using the M600 macro you have, because you may need to move your save state there otherwise you may be capturing the G91 from your UNLOAD Macro.  I get a little uncertain about the order things happen in, but saving the state at the get go might help.

boardthatpowder, 2020-01-22:
Didn’t know about the MOVE=1.  Thanks for pointing that out.  Will give that a try. After reading the docs looks like I’ll have to remove RESUME and replace with CLEAR_PAUSE too as that manages its own state.

Just read in the docs about the pattern about wrapping all G1’s with save and restore commands. Will try that too.

Will also doublecheck the UNLOAD macro.  Will try all these out tomorrow and report back.

boardthatpowder, 2020-01-22:
Seem to be having more success with the following changes.  Tried 10 times, all resumed ok.  Main difference is the removal of the `RESTORE_GCODE_STATE` I had in my resume macro and instead rely on the standard PAUSE/RESUME macros to manage state.  Will give it a more thorough testing before closing.

```
# Filament change
[gcode_macro M600]
gcode:
    PAUSE_MACRO
    UNLOAD

[gcode_macro PAUSE_MACRO]
gcode:
    PAUSE
    RESPOND TYPE=command MSG=action:paused
    PARK_MACRO
    SET_IDLE_TIMEOUT TIMEOUT=3600

[gcode_macro RESUME_MACRO]
gcode:
    SET_IDLE_TIMEOUT TIMEOUT=600
    RESUME
    RESPOND TYPE=command MSG=action:resumed

[gcode_macro PARK_MACRO]
default_parameter_X: 20
default_parameter_Y: 280
default_parameter_Z: 100
gcode:
    SAVE_GCODE_STATE NAME=PARK_MACRO_state
    G91                     ; relative positioning
    G1 E-2 F1000            ; retract filament
    G1 Z10                  ; lift z slightly             
    G90                     ; absolute positioning
    G1 X{X} Y{Y} Z{Z} F3000 ; park the head
    RESTORE_GCODE_STATE name=PARK_MACRO_state
```

KevinOConnor, 2020-02-14:
I'm going to close this as it looks like a solution was found for the original issue.

-Kevin

closed 2020-02-14 by KevinOConnor

MrSurly, 2021-04-07:
@boardthatpowder Did that ultimately work for you?