Klipper3d/klipper #2413: Virtual CD print and M600 macro

ZhenyaKa, 2020-01-22:
Is it possible to pause and change filament when printing from virtual CD?

I got  error:
---
M98 P"0:/macros/M600"
Could not run code because a bad response has been received
---

FHeilmann, 2020-01-22:
M98 indicates you're running external code (DWC2 for klipper) which is not part of this repository. I suggest you file this bug report with the dwc2-for-klipper guys instead.

ZhenyaKa, 2020-01-22:
> M98 indicates you're running external code (DWC2 for klipper) which is not part of this repository. I suggest you file this bug report with the dwc2-for-klipper guys instead.

I run M600 via command line and have the same error. But if printer not printing (just hommed and heated) M600 works fine.

FHeilmann, 2020-01-22:
klipper does not officially support M600 (note this gcodes absence from the list of available G-Codes here: https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

So unless you define a gcode macro like shown here: https://github.com/KevinOConnor/klipper/blob/master/config/sample-macros.cfg#L103
klipper will simply dismiss M600 calls (made via buttons or commandline).

If you're using this macro and still get an error, please attach your `klippy.log`. The error message you posted is generated by the dwc2 addon and doesn't allow us to troubleshoot that error here. `klippy.log` is required for these kinds of errors.

Bests,
Florian

ZhenyaKa, 2020-01-22:
I have M600 macro. It runs fine when printer not printing!
Homed, heated, M600 - runs okay

But when printing I can't run it with error.

Maybe you can tell me how to change filament via LCD or gcode command when printing?
Not placing command in gcode, but run that command manually.

For example I print big part ant filament ending. 
How to replace filament? (in virtual sd-card mode)

FHeilmann, 2020-01-22:
Please share your `klippy.log` from when the error occured. Otherwise debugging this issue is near impossible.

closed 2020-01-29 by klipper-gitissuebot