Klipper3d/klipper #2414: Pause macro with headup

ZhenyaKa, 2020-01-22:
Printing with dwc2.

When pressing pause - head freeze.

Need to up it a little:
[gcode_macro MOVE_UP]
gcode:
  SAVE_GCODE_STATE NAME=my_move_up_state
  G91
  G1 Z10 F2000
  RESTORE_GCODE_STATE NAME=my_move_up_state

Is it possible to add this code to macro "pause"?

ZhenyaKa, 2020-01-22:
[gcode_macro MOVE_UP]
gcode:
  SAVE_GCODE_STATE NAME=my_move_up_state
  G91
  G1 Z10 F3000
  RESTORE_GCODE_STATE NAME=my_move_up_state

[gcode_macro PAUSE_PRINT]
gcode:
 MOVE_UP

Here my solution, but it prints in air one layer and then lower head only :(

ld3300, 2020-01-24:
try `RESTORE_GCODE_STATE NAME=my_move_up_state MOVE=1`

closed 2020-01-29 by klipper-gitissuebot

ZhenyaKa, 2020-01-29:
Why you need log for that question???