Klipper3d/klipper #2415: Klipper Stops with Simolify3D Gcode

laubed, 2020-01-22:
Hi,

First, Klipper is awesome. I finally achieve the speed that i hope to achieve. Communincation with tmc drivers is working reliable. Never thought that the improvements were that noticable with just a firmware change.
But i have a serious issue: Simplify3D outputs gcode with some extremely high filament extrusion values. On some prints it wont happen, on others a few hours into the print.

Looking up the error klipper throws it shows that this issue was reported already. 

I would like a config option that ignores the E part of that move or change it to max extrusion volume. That way simplify3d users can print reliably.

Thank you in advance.

FHeilmann, 2020-01-22:
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L107

setting this to a large value will achieve what you're looking for.
However I would suggest getting to the bottom of the issue and figuring out which particular behavior of S3D causes the error. By increasing the config parameter you're merely treating the symptoms.

-Florian

laubed, 2020-01-22:
I have increased the mentioned config parameter. Now instead of stopping it extrudes a blob of plastic.
I totally agree that the real issue is simplify3d. I guess it is kind of the same problem that smoothieware had with s3d. I just want klipper to ignore these gcodes and not fail the prints.

This evening i will write a programm to analyze the mentioned gcode to get an understanding of what s3d thinks or intends.

FHeilmann, 2020-01-22:
Unfortunately ignoring such g-codes is not feasible since they are likely going to be 'G1' moves. What I would suggest is turning off the advanced extrusion features in S3D (coasting, wiping, etc. pp) and try to figure out which feature causes these malicious G1 moves. 

I suspect this should be fixable, since this issue tracker would otherwise be overrun with S3D users reporting similar problems.

laubed, 2020-01-22:
i have no extra extrusion features enabled except retraction.

Boffgithub, 2020-01-22:
Whenever I had this problem I always tracked it down to a bridging feature.  As FHeilmann has said, changing the config isn't really fixing the issue.
In S3D, have a look in the preview to see if there is any bridging. Try reducing the bridging settings or turn it off completely.  Also try disabling 'Apply bridging settings to perimeters'

laubed, 2020-01-22:
i've written a small python program to analyze gcode files. It prints the 10 largest Extrusion/Distance values of a gcode file.

`
import math

f = open("Gantry_Spacer_F.gcode")

E = 0
X = 0
Y = 0
Evals = []
lineNo = 0

for line in f:
    line = line.strip()
    lineNo += 1

    segments = line.split(" ")

    if(segments[0] == "G92" and segments[1] == "E0.0000"):
        E = 0

    if(segments[0] == "G1"):
        value = None
        x = None
        y = None
        for i in segments:
            if( i.startswith("E") ):
                value = float(i[1:])

            if( i.startswith("X") ):
                x = float(i[1:])

            if( i.startswith("Y") ):
                y = float(i[1:])

        if( value != None and x != None and y != None):
            diff = value - E
            dst = math.sqrt((x-X)*(x-X) + (y-Y)*(y-Y))
            if(dst != 0):
                diff = diff / dst
                Evals.append((diff, lineNo))
            E = value
            X = x
            Y = y

finished = sorted(Evals, reverse=True)[0:10]

for i in finished:
    print(i)
`

This is the output of a print that succeeded without any fixes:
`(0.23001387271253482, 179349)
(0.11815768609520728, 5199)
(0.11780644859286588, 182535)
(0.11532894411275056, 6211)
(0.10580078696435072, 181542)
(0.10090629038702023, 183402)
(0.06251004603962797, 3210)
(0.05883484054148938, 1538)
(0.05854905538444194, 1542)
(0.05820855000873391, 2340)`

And one that failed:
`(18.869022976055792, 8230)
(17.078719413579304, 8211)
(4.049703426997204, 8175)
(3.0037286050161476, 8173)
(2.9113744443668574, 8192)
(2.112103584671983, 8174)
(1.2869502053059356, 8172)
(1.0957241509106959, 8309)
(1.0945815415130893, 8295)
(1.086183056357549, 8247)`

That clearly shows that something is indeed wrong. I also sliced a model 3 times: one with my default settings, one with bridging turned off for perimeters and one with default bridging settings. All resulted in the same values (order of magnitude).

I will investigate that further

dmbutyugin, 2020-01-22:
Note that you can ssh to your RaspberryPi, re-enable the check you may have disabled earlier (https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L107) and run the following command:
```
~/klippy-env/bin/python ~/klipper/klippy/klippy.py ~/printer.cfg -d ~/klipper/out/klipper.dict -i /path/to/troublesome.gcode -o /dev/null
```
This is a dry-run mode, so Klipper will process the gcode file (and crash if it encounters bogus code) but will not print anything. This way you can quickly test various versions of gcode with different S3D parameters to see which one can be causing issues without having to run many hours prints.

krpepe, 2020-01-23:
I use s3d and have never had this issue. Just a guess but possibly something in the gcode script that is causing this?Sent via the Samsung Galaxy S8 Active, an AT&T 5G Evolution capable smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/KevinOConnor/klipper/issues/2415?email_source=notifications\u0026email_token=AGMAWGGTPA47YYDG7S27NGLQ7BOO7A5CNFSM4KKBFGK2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJT5OEY#issuecomment-577230611",
"url": "https://github.com/KevinOConnor/klipper/issues/2415?email_source=notifications\u0026email_token=AGMAWGGTPA47YYDG7S27NGLQ7BOO7A5CNFSM4KKBFGK2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJT5OEY#issuecomment-577230611",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

laubed, 2020-01-23:
I can now say that it is definitly the option apply bridging settings to perimeters.
Without the option turned on my script outputs:

    (0.6349380290232707, 68232)
    (0.4224830776287092, 70802)
    (0.41684225598286373, 76155)
    (0.41481734564845524, 160519)
    (0.4093934786812962, 155134)
    (0.4075855230255742, 81364)
    (0.4009322462125237, 149805)
    (0.39953051544372326, 86711)
    (0.39359653568165387, 144575)
    (0.31120404709344784, 62588)

with the option turned on i get these insane high values:

    (17.137005494580652, 7172)
    (15.175013541004564, 7153)
    (3.605831351832589, 7117)
    (2.8536678017445363, 7134)
    (2.689439405368033, 7115)
    (1.9131796117186843, 7116)
    (1.1761723050758215, 7114)
    (1.0750315740945677, 7251)
    (1.0747367667191854, 7237)
    (1.0639445469401032, 7189)

looking up that line in the gcode shows this move sequence mid print:

    G1 X133.382 Y121.607 E10.4017
    G1 X133.480 Y121.580 E12.1437

that are nearly 2mm of extrusion on a few tenth of a milimeter movement. Doing these moves results in a large blob... interestingly this section is not anywhere near a bridging section.

i will now contact simplify3d support with the evidence found.

Boffgithub, 2020-01-24:
Glad you found the root cause of the issue.  This was actually the setting that I found was affecting it too for me, although mine was done through trial and error!  Since leaving it disabled I haven't had any more max extrusion errors, and that includes slice mdoels with bridging in them.

alfsoft, 2020-01-26:
I went another way. I simply modify extruder.py to disable this error every time I update Klipper. Described what I did here: [https://github.com/KevinOConnor/klipper/issues/2366#issuecomment-578521470](https://github.com/KevinOConnor/klipper/issues/2366#issuecomment-578521470)

laubed, 2020-01-26:
> I went another way. I simply modify extruder.py to disable this error every time I update Klipper. Described what I did here: [#2366 (comment)](https://github.com/KevinOConnor/klipper/issues/2366#issuecomment-578521470)

while possible even that solution causes an overextrusion at that point. Still better than a broken print

closed 2020-01-29 by klipper-gitissuebot