Klipper3d/klipper #2416: Unable to init TMC5160 on RE-ARM/RAMPS1.6+ (possible cs_pin issue)

BoHoehne, 2020-01-22:
Hello,

i have a RAMPS 1.6+ on a Re-ARM board. 
The SPI ports are hard wired on the PCB to SDK,SCK and SDO (jumpers where double checked). 
CS is wired with a separate cable (ohm tested on all cables and they are OK).
Diag1 is also wired with a separate cable (tested OK as well).
There is no display or z-sensor attached.

All the documents about re-arm combined with RAMPS 1.6+ say that the SDK,SCK and SDO are actually wired to the Re-ARMs SPI bus.

For testing i issued the following command sequence (mind the wildcards):

init_tmc stepper=stepper_(a-c)/extruder(1)
dump_tmc stepper=stepper_(a-c)/extruder(1)
they alway return something similar to the following:

Recv: // ========== Write-only registers ==========
Recv: // TPWMTHRS:   00000006 TPWMTHRS=6
Recv: // COOLCONF:   00030000 sgt=3
Recv: // IHOLD_IRUN: 00060704 IHOLD=4 IRUN=7 IHOLDDELAY=6
Recv: // PWMCONF:    c40d001e PWM_OFS=30 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000000
Recv: // CHOPCONF:   00000000 MRES=0(256usteps)
Recv: // GSTAT:      00000000
Recv: // DRV_STATUS: 00000000
Recv: // FACTORY_CONF: 00000000
Recv: // IOIN:       00000000 VERSION=0x0
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      00000000
Recv: // MSCURACT:   00000000
Recv: // OTP_READ:   00000000
Recv: // PWM_SCALE:  00000000
Recv: // PWM_AUTO:   00000000
Recv: // TSTEP:      00000000

What did i miss ?
Any hints are welcome.

[klippy_log.zip](https://github.com/KevinOConnor/klipper/files/4098732/klippy_log.zip)

[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/4098784/printer_cfg.zip)

laubed, 2020-01-22:
The comments in the lcd example files state that some controllers need a specific spi bus to be specified:
https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg

perhaps that is also the case for other spi periphals like your tmc drivers.

BoHoehne, 2020-01-22:
Thanks,

the SPI bus name is not documented by pannucat, but i tried SW_SPI by basically overwriting the existing SPI ports with software ones. Like:

spi_software_sclk_pin: P0.15
spi_software_mosi_pin: P0.18
spi_software_miso_pin: P0.17

Sadly, it didn't work ether.
I was hoping to be able to use the existing PCB wiring for the SPI ports. 
Still stuck.

BoHoehne, 2020-01-24:
This problem is linked to issue #2424

jalanjarosz, 2020-01-24:
You can try adding spi_bus: <MCU name> before the spi_software options 

I had to add a i2c_mcu option when testing a ssd1306 display on my Re-ARM.  
I also tested a Fysetc MKS Mini v2.1, which I believe uses SPI for the display, and did not have to specify SPI pins or MCU.

KevinOConnor, 2020-02-02:
Is this issue resolved now that #2424 is?

-Kevin

KevinOConnor, 2020-02-14:
Okay, I'm going to close this as a dup of #2424. 

-Kevin

closed 2020-02-14 by KevinOConnor