Klipper3d/klipper #2419: No Flymaker Flyboard config

DJDOUBLEG, 2020-01-22:
HI!

I make config file for this board.
https://www.aliexpress.com/item/4000531805672.html

[generic-flymaker-flyf407zg.cfg.zip](https://github.com/KevinOConnor/klipper/files/4100058/generic-flymaker-flyf407zg.cfg.zip)

Please check it.

Thank you!

https://www.facebook.com/groups/Flyboard32bitprinterboard/

DJDOUBLEG, 2020-01-22:
Dear Issue bot... I don't have at this moment log file. :)
This project under construction.

jakep82, 2020-01-23:
This should be submitted as a pull request.

KevinOConnor, 2020-01-24:
Okay, thanks.  I added a config for this board (commit 8bf3e563).

-Kevin

AniInABox, 2020-01-27:
@DJDOUBLEG do you have klippy successfully installed to the FlyBoard?

DJDOUBLEG, 2020-01-27:
> @DJDOUBLEG do you have klippy successfully installed to the FlyBoard?

Hi!

Sorry.
Not yet.
I don't have time now.
Only i make pin config, based on fly maker (flyboard) marlin config and based on (https://github.com/FLYmaker/FLYF407ZG/blob/master/picture/Pin%20diagram.png?raw=true) this pin out pinture.
 
My opinion is same config with SKR pro (same CPU).
Maybe boot loader is the different.

KevinOConnor, 2020-02-14:
Okay, I'm going to close this as an example config was added as part of  8bf3e56.

-Kevin

closed 2020-02-14 by KevinOConnor